meshtastic/firmware #3173: [Bug]: I can't activate GPS

Ronda-00, 2024-02-06:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.21.7

### Description

I activate the option within position
 and when restarting the t-beam
  the GPS is deactivated again. thanks

jp-bennett, 2024-02-07:
This is probably due to a change to firmware, that the mobile clients are still rolling out.

Ronda-00, 2024-02-07:
I have installed firmware 2.2.20 and the GPS works perfectly

BrunoKlu, 2024-02-08:
Have same issue with all my nodes on 2.21 will try with android app but same issue on Ios and web client

mverch67, 2024-02-11:
Until the mobile clients have implemented the new gpsMode you could use the CLI to activate the GPS.
`meshtastic --set gps_mode ENABLED`

jp-bennett, 2024-02-11:
> Until the mobile clients have implemented the new gpsMode you could use the CLI to activate the GPS.
> `meshtastic --set position.gps_mode ENABLED`

Unfortunately pip seems to be only offering 2.20 at this time.

mverch67, 2024-02-16:
In the meantime GPS_mode is supported by CLI, Android and Apple app... can you re-test and close this report if everything works?

closed 2024-02-20 by thebentern