meshtastic/firmware #3174: [Board]: LilyGo Lora V1.4

guerradr, 2024-02-06:
### SOC

ESP32

### Lora IC

SX1278

### Product Link

https://www.aliexpress.us/item/2251832638444203.html?gatewayAdapt=glo2usa4itemAdapt

### Description

[image]

guerradr, 2024-02-07:
#define I2C_SDA 21 // I2C pins for this board
#define I2C_SCL 22

// #define RESET_OLED 16 // If defined, this pin will be used to reset the display controller

// #define VEXT_ENABLE 21 // active low, powers the oled display and the lora antenna boost
// #define LED_PIN 2      // If defined we will blink this LED
#define BUTTON_PIN 0   // If defined, this will be used for user button presses
// #define BUTTON_NEED_PULLUP
// #define EXT_NOTIFY_OUT 13 // Default pin to use for Ext Notify Module.

#define BATTERY_PIN 35 // A battery voltage measurement pin, voltage divider connected here to measure battery voltage
#define ADC_CHANNEL ADC1_GPIO35_CHANNEL

#define USE_RF95    // RFM95/SX127x
#define LORA_DIO0 26  // a No connect on the SX1262/SX1268 module
#define LORA_RESET 4 // RST for SX1276, and for SX1262/SX1268
#define LORA_DIO1 33  // IRQ for SX1262/SX1268
#define LORA_DIO2 32  // BUSY for SX1262/SX1268
#define LORA_DIO3     // Not connected on PCB, but internally on the TTGO SX1262/SX1268, if DIO3 is high the TXCO is enabled

#undef LORA_SCK
#define LORA_SCK 18
#undef LORA_MISO
#define LORA_MISO 19
#undef LORA_MOSI
#define LORA_MOSI 23
#undef LORA_CS
#define LORA_CS 5

// RX/TX for RFM95/SX127x
// #define RF95_RXEN LORA_RXEN
// #define RF95_TXEN LORA_TXEN
// #define RF95_TCXO <GPIO#>

// #undef GPS_RX_PIN
// #undef GPS_TX_PIN
// #define GPS_RX_PIN 37
// #define GPS_TX_PIN 38 // per @eugene

thebentern, 2024-03-28:
Since this is a niche board with an SX127X based radio instead of an SX126X based one, it doesn't provide much value considering there are better alternatives from LilyGo like the T3S3, T-Deck, T-Beam Supreme that all have the superior radio.

closed 2024-03-28 by thebentern