meshtastic/firmware #3187: [Bug]: Heltec Tracker display reacts with a delay and the display does not go out

spacepc-de, 2024-02-09:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.2.19

### Description

I have 7 Heltec trackers with the current firmware 2.2.19. On all of them, the display does not go out after a defined timeout, no matter what I set. In addition, when I press the button to change the views, it reacts with a long delay or not at all. Since the display is always on, the battery life is greatly reduced.

Gabrielerusso, 2024-02-11:
> The button accuracy should be improved by this PR: #3188. The screen is (re)printed one time/sec, so it takes one second to advance to the next.
> 
> The newer heltec tracker model v1.1 has a different power logic. AFAIK it's not possible to switch off the screen without also switching off the LoRa chip.

It is possible to switch the screen off, it was the 1.0 that has a problem. My PR #3163 fixes it.

BobsBlueNorth, 2024-02-12:
I have been seeing the same issue on my v1.1 boards with all newer firmware since 2.11. By rolling back to 2.11 they are useable. I just tried 2.22 today but no dice. The screen "locks" up at the timeout.

I didn't realize there was a 2.23 to try.

dchauran, 2024-03-04:
Came here to report thr.timeout issue, only to find that it had not only been reported but fixed and was in Alpha. I installed alpha and it worked. Screen refresh is faster and the display does indeed turn off now.

closed 2024-03-04 by thebentern