meshtastic/firmware #3190: [Bug]: SSL - Memory allocation failed

tavancini, 2024-02-09:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.19

### Description

Hardware: T-Beam v1.1 SX1268 433/470Mhz

SSL error occurs when trying to connect to a public MQTT server in Brazil:

Host: platform.meshbrasil.com,
Port: 1883
Username: meshdev
Password: large4cats
Root topic: meshdev

### Relevant log output

```Shell
INFO  | 16:44:51 14340 [mqtt] Using TLS-encrypted session
INFO  | 16:44:51 14340 [mqtt] Attempting to connect directly to MQTT server platform.meshbrasil.com, port: 1883, username: meshdev, password: large4cats
[14340499][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[14340500][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():264]: (-32512) SSL - Memory allocation failed
```

thebentern, 2024-02-12:
This seems like an SSL certificate issue rather than an issue with the firmware

closed 2024-02-12 by thebentern

oherrala, 2024-04-20:
This might be related to #3496.