meshtastic/firmware #3196: Signal measurement on screen does not go above 86%

5kamisama5, 2024-02-10:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.2.19.8f6a283 Beta

### Description

The Heltec V3 never gets a signal above 86% even when touching the antennas together it's the highest I've ever seen it. I've used multiple antennas of various sizes and prices and across over 15 nodes and nothing higher is seen on the screen of any of the V3s. I can however see 100% signal strength on the Heltec V2s. I've confirmed all nodes are on the same firmware version, same channel and lora Channel frequency(20). I've exhausted all other options. 
The one on top is V2 and the bottom one is V3.
[image]

GUVWAF, 2024-02-10:
It's a pretty simple calculation which I would take with a grain of salt: 
https://github.com/meshtastic/firmware/blob/1085b54069e426b592f7d458548de6e485f72c8b/src/graphics/Screen.cpp#L833
So it seems you're not getting an SNR higher than 7dB. The V2 has a SX127x radio, while the V3 has a SX126x. Maybe if you turn off Rx boosted gain on the V3, it will be better in this case. 
SNR may actually be worse if the devices are so close.

5kamisama5, 2024-02-10:
I just turned off the Rx boosted on both sender and receiving node and went about 20ft from the node and the signal strength remains unchanged.

closed 2024-03-02 by thebentern