meshtastic/firmware #3201: [Feature Request]: Disable MQTT when/while RangeTest enabled

RightHandMan1, 2024-02-10:
### Platform

NRF52, ESP32

### Description

Was thinking it would be nice to make Range Test enable switch, to automatically disable MQTT while RT is enabled. When RT is disabled, MQTT would be re-enabled(if previously on). Believe doing this will stop most of the RT messaging to others' nodes and MQTT, and utilizing the extra airtime especially in time limited areas and processing. This could also allow removal of the 'dont process RT seq' code in the apps. I cant think of a good reason to have the RT traffic going over MQTT, let alone all others' nodes. At least, put a blurb in the docs and apps RT screen, to disable MQTT before you RT...

RightHandMan1, 2024-02-12:
@thebentern  ??

andrekir, 2024-03-13:
we have a few safeguards to prevent range test spamming, and one of them is blocking the sequence messages from reaching the public MQTT server. so this is already covered.

closed 2024-03-14 by thebentern