meshtastic/firmware #3207: [Feature Request]: Spread telemetry/GPS updates across channels

tropho23, 2024-02-11:
### Platform

NRF52

### Description

(This feature request is *not* only for nrf52 but the platform choice does not allow me to choose 'all platforms')

Request firmware feature to spread telemetry/GPS updates across multiple channels.

Currently users are limited to using primary channel for sending telemetry and GPS data.  Ideally, selective send of this data to any channel(s) we choose but this has been limited to primary channel to avoid mesh saturation/channel utilization.

If allow sending GPS updates to any/all channels but limit update intervals per channel to respect the *total* interval update that would be rather useful.  For example:

_(With GPS update interval set to 10 minutes)_
00:00=send GPS update on channel 0/primary
00:10=send GPS update on channel 1
00:20=send GPS update on channel 2
00:30=send GPS update on channel 3
00:40=send GPS update on channel 0/primary (40 minutes since last update)
...and so on.  The idea is not exceed the GPS update interval in total, which is 10 minutes.

The example above allows users to send GPS updates via more than only primary channel, but updates respect the 10 minute interval set.  This allows channel utilization to remain the same as if GPS updates are sent only on the primary channel.

I recognize users can increase GPS update interval and flood the mesh but they can already do that so this proposal adds zero risk if implemented in the manner suggested above.

This feature cannot be depending on mobile apps to function as standalone devices should be able to use this feature.

jp-bennett, 2024-02-11:
Somewhat related is the desire to set precision levels for different channels. Possibly through a bitmask that gets AND'd with the location at https://github.com/meshtastic/firmware/blob/master/src/modules/PositionModule.cpp#L110 before sending the packet.

garthvh, 2024-02-11:
Seems really complex, uses a bunch of airtime end is of marginal benefit, still not sure why secondary channels need position. Also has to be done in 3.0.

andrekir, 2024-02-11:
we really don't want to send the same packet multiple times in different channels. this would have a huge impact in total airtime considering rebroadcasts/ACKs.

GUVWAF, 2024-02-11:
I agree with Garth and Andre. The suggestion to let it send on a different channel at each opportunity is good, but then again if you want frequent updates on one channel, you’ll need to set the updates really high. Duplicating the settings for each channel seems very cumbersome.

Indeed someone can already spam the channel by setting the position updates to every 5 seconds, but it’s pretty clear that that creates a lot of traffic. However, it’s not intuitive that enabling it on an additional channel means that for each update your device will need to send a separate packet there. 

Allowing someone to _request_ position with less precision on secondary channels might be an alternative solution.

jp-bennett, 2024-02-11:
> Seems really complex, uses a bunch of airtime and is of marginal benefit, still not sure why secondary channels need position. Also has to be done in 3.0.

Adds complexity, sure. The idea is to round-robin the location data among the channels where it's enabled, so no additional airtime.

For use case, I want my family to know exactly where I am, but I only want the world at large to see approximate location.

Why would it need to be 3.0? A receiving device has no clue if a position was sent over a secondary channel. All we need is added config on the protobuf, and some code to send out messages on different channels.

GUVWAF, 2024-02-11:
> For use case, I want my family to know exactly where I am, but I only want the world at large to see approximate location.

Assuming you don't want the whole world to be able to follow you in real-time, wouldn't it be better to allow manual imprecise position requests on secondary channels? I'd argue that it's an even better user experience, because you will get a response immediately, instead of having to wait for the next broadcast opportunity, whose interval you don't know.

jp-bennett, 2024-02-11:
> Assuming you don't want the whole world to be able to follow you in real-time, wouldn't it be better to allow manual imprecise position requests on secondary channels?

Does that work for MQTT? Another use case for this is to let people send their location data to a public MQTT server, to have a more accurate map of where nodes are at. Local trusted mesh gets full accuracy location, public channel gets reduced accuracy, and is shared with MQTT.

I think we have enough agreement to get started. Whether to enable the automatic position broadcast on a secondary channel is a relatively minor implementation detail. For that matter, it could be disabled by default, and just turned on for custom firmware builds.

tropho23, 2024-02-11:
It's already been mentioned I will reiterate that the intent is **not** to broadcast this information on multiple channels and increase utilization, it's to broadcast _at the same interval_ but spread among the selected channels.  No additional position updates will be sent out; please review the example I provided in the original summary.

thebentern, 2024-02-11:
We do need to think about better control mechanisms for which portnums can communicate over which channels IMO, because we have the same problem w/ MQTT. For instance, users want to be able to send nodeinfo, telemetry, or position over secondary uplink channels and those don't necessarily have the same affect on airtime. 

I had originally envisioned that being an enum w/ flags for portnums on each channel with mqtt uplink. Not trying to highjack this issue, just noting that there are similarities in the two in the fact that users want more control over _what_ information goes out over secondary meshes, be that lora or mqtt.

jp-bennett, 2024-02-12:
> I had originally envisioned that being an enum w/ flags

We could do that with the GPS stuff, too. Maybe define high, mid, and low accuracy GPS flags.

closed 2024-07-03 by garthvh