meshtastic/firmware #3212: Build failed version 2.2.23 for env tbeam-s3-core

ric-tim, 2024-02-11:
Problem with ets_sys.h  ?

In file included from C:/Users/richa/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120,
                 from C:/Users/richa/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/Users/richa/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:32,
                 from src/mesh/MemoryPool.h:3,
                 from src/mesh/MeshRadio.h:3,
                 from src/airtime.h:3,
                 from src/airtime.cpp:1:
C:/Users/richa/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32/rom/ets_sys.h:21:10: fatal error: soc/soc.h: No such file or directory
 #include "soc/soc.h"
          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\tbeam-s3-core\src\airtime.cpp.o] Error 1

mverch67, 2024-02-12:
It looks like your development environment is out of date or corrupt. Try to remove all the files in your home folder under
.platformio/packages/framework-arduinoespressif32*
.platformio/packages/toolchain-xtensa-esp32*
.platformio/packages/tool-openocd-esp32*
.platformio/packages/platforms/espressif32*

Don't worry, platformio will automatically download and re-create these folders.

ric-tim, 2024-02-12:
Thanks a lot : tbeam-s3-core  SUCCESS   00:11:40.175

closed 2024-02-12 by ric-tim