meshtastic/firmware #3218: [Bug]: RAK4631 Shutting down after updating Firmware to 2.2.21.7f7c5cb

vinodcombat, 2024-02-12:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.2.21.7f7c5cb

### Description

The device keeps shutting down and restarting after erasing the flash with nrf_erase2 and flashing latest firmware. I could replicate the same issue on two different hardware.

### Relevant log output

```Shell
@INFO  | ??:??:?? 4 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 4 Filesystem files:
DEBUG | ??:??:?? 4  db.proto (127 Bytes)
DEBUG | ??:??:?? 4  channels.proto (53 Bytes)
DEBUG | ??:??:?? 4  config.proto (87 Bytes)
DEBUG | ??:??:?? 4  module.proto (113 Bytes)
DEBUG | ??:??:?? 4 Using analog input 5 for battery level
INFO  | ??:??:?? 4 Scanning for i2c devices...
DEBUG | ??:??:?? 4 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 4 I2C device found at address 0x3c
INFO  | ??:??:?? 4 ssd1306 display found
INFO  | ??:??:?? 4 ssd1306 display found
DEBUG | ??:??:?? 4 0x3 subtype probed in 2 tries 
INFO  | ??:??:?? 4 1 I2C devices found
DEBUG | ??:??:?? 4 acc_info = 0
INFO  | ??:??:?? 4 Meshtastic hwvendor=9, swver=2.2.21.7f7c5cb
DEBUG | ??:??:?? 4 Reset reason: 0x0
DEBUG | ??:??:?? 4 Setting random seed 1445246187
INFO  | ??:??:?? 4 Initializing NodeDB
INFO  | ??:??:?? 4 Loading /prefs/db.proto
INFO  | ??:??:?? 4 Loaded saved devicestate version 22
INFO  | ??:??:?? 4 Loading /prefs/config.proto
INFO  | ??:??:?? 4 Loaded saved config version 22
INFO  | ??:??:?? 4 Loading /prefs/module.proto
INFO  | ??:??:?? 4 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 4 Loading /prefs/channels.proto
INFO  | ??:??:?? 4 Loaded saved channelFile version 22
INFO  | ??:??:?? 4 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 4 cleanupMeshDB purged 0 entries
WARN  | ??:??:?? 4 Using nodenum 0x475bcb9e 
DEBUG | ??:??:?? 4 Expanding short PSK #1
INFO  | ??:??:?? 4 Wanted region 10, using IN
DEBUG | ??:??:?? 4 region=10, NODENUM=0x475bcb9e, dbsize=1
INFO  | ??:??:?? 4 Setting GPS power=1
DEBUG | ??:??:?? 4 WANT GPS=1
INFO  | ??:??:?? 4 Setting GPS power=1
DEBUG | ??:??:?? 4 NeighborInfoModule is disabled
INFO  | ??:??:?? 4 No /prefs/ringtone.proto preferences found
INFO  | ??:??:?? 4 Initializing External Notification Module
INFO  | ??:??:?? 4 Using Pin 36 in digital mode
INFO  | ??:??:?? 4 Turning on screen
DEBUG | ??:??:?? 4 Module wants a UI Frame
WARN  | ??:??:?? 4 SX126xInterface(cs=42, irq=47, rst=38, busy=46)
DEBUG | ??:??:?? 4 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | ??:??:?? 4 Starting meshradio init...
DEBUG | ??:??:?? 4 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 4 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 4 Radio freq=865.875, config.lora.frequency_offset=0.000
[... 107 lines cut ...]
DEBUG | ??:??:?? 49 [DeviceTelemetryModule] nRF52 encrypt fr=475bcb9e, num=42b376a1, numBytes=28!
DEBUG | ??:??:?? 49 [DeviceTelemetryModule] decoded message (id=0x42b376a1 fr=0x9e to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | ??:??:?? 49 [RadioIf] Starting low level send (id=0x42b376a1 fr=0x9e to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
DEBUG | ??:??:?? 49 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=44, time 559 ms
DEBUG | ??:??:?? 49 [RadioIf] AirTime - Packet transmitted : 559ms
DEBUG | ??:??:?? 50 [RadioIf] Completed sending (id=0x42b376a1 fr=0x9e to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
INFO  | ??:??:?? 53 [Button] Shutdown from long pressshowing shutdown screen
DEBUG | ??:??:?? 63 [Button] Long press stop!
INFO  | ??:??:?? 66 [Button] Turning off screen
INFO  | ??:??:?? 66 [Button] Shutting down
INFO  | ??:??:?? 66 [Button] Entering deep sleep forever
INFO  | ??:??:?? 66 [Button] Disable NRF52 bluetooth
DEBUG | ??:??:?? 66 [Button] SX126x entering sleep mode (FIXME, don't keep config)
INFO  | ??:??:?? 66 [Button] GPS deep sleep!
DEBUG | ??:??:?? 66 [Button] WANT GPS=0
DEBUG | ??:??:?? 66 [Button] GPS Lock took 61, average 0
INFO  | ??:??:?? 66 [Button] Setting GPS power=0
DEBUG | ??:??:?? 66 [Button] Sleep Time: 120000
INFO  | ??:??:?? 66 [Button] Saving /prefs/db.proto
INFO  | ??:??:?? 67 [Button] Saving /prefs/config.proto
INFO  | ??:??:?? 68 [Button] Saving /prefs/module.proto
INFO  | ??:??:?? 68 [Button] Saving /prefs/channels.proto
INFO  | ??:??:?? 69 [Button] Setting GPS power=0
DEBUG | ??:??:?? 6
```

mverch67, 2024-02-12:
I would be interested if this also happens in 2.2.22. Could you check please?

vinodcombat, 2024-02-12:
> I would be interested if this also happens in 2.2.22. Could you check please?

The same issue persists with 2.2.22.404d0dd.

Log
@INFO  | ??:??:?? 1 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1  db.proto (118 Bytes)
DEBUG | ??:??:?? 2  channels.proto (53 Bytes)
DEBUG | ??:??:?? 2  config.proto (87 Bytes)
DEBUG | ??:??:?? 2  module.proto (112 Bytes)
DEBUG | ??:??:?? 2 Using analog input 5 for battery level
INFO  | ??:??:?? 2 Scanning for i2c devices...
DEBUG | ??:??:?? 2 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 2 I2C device found at address 0x3c
INFO  | ??:??:?? 2 ssd1306 display found
INFO  | ??:??:?? 2 ssd1306 display found
DEBUG | ??:??:?? 2 0x3 subtype probed in 2 tries 
INFO  | ??:??:?? 2 1 I2C devices found
DEBUG | ??:??:?? 2 acc_info = 0
INFO  | ??:??:?? 2 Meshtastic hwvendor=9, swver=2.2.22.404d0dd
DEBUG | ??:??:?? 2 Reset reason: 0x0
DEBUG | ??:??:?? 2 Setting random seed 3412343872
INFO  | ??:??:?? 2 Initializing NodeDB
INFO  | ??:??:?? 2 Loading /prefs/db.proto
INFO  | ??:??:?? 2 Loaded saved devicestate version 22
INFO  | ??:??:?? 2 Loading /prefs/config.proto
INFO  | ??:??:?? 2 Loaded saved config version 22
INFO  | ??:??:?? 2 Loading /prefs/module.proto
INFO  | ??:??:?? 2 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 2 Loading /prefs/channels.proto
INFO  | ??:??:?? 2 Loaded saved channelFile version 22
INFO  | ??:??:?? 2 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 2 cleanupMeshDB purged 0 entries
WARN  | ??:??:?? 2 Using nodenum 0xba1a41d3 
DEBUG | ??:??:?? 2 Expanding short PSK #1
INFO  | ??:??:?? 2 Wanted region 10, using IN
DEBUG | ??:??:?? 2 region=10, NODENUM=0xba1a41d3, dbsize=1
INFO  | ??:??:?? 2 Setting GPS power=1
DEBUG | ??:??:?? 2 WANT GPS=1
INFO  | ??:??:?? 2 Setting GPS power=1
DEBUG | ??:??:?? 2 NeighborInfoModule is disabled
INFO  | ??:??:?? 2 No /prefs/ringtone.proto preferences found
INFO  | ??:??:?? 2 Initializing External Notification Module
INFO  | ??:??:?? 2 Using Pin 36 in digital mode
INFO  | ??:??:?? 2 Turning on screen
DEBUG | ??:??:?? 2 Module wants a UI Frame
WARN  | ??:??:?? 2 SX126xInterface(cs=42, irq=47, rst=38, busy=46)
DEBUG | ??:??:?? 2 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 2 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 2 Radio freq=865.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=IN, name=LongFast, config=0, ch=3, power=30
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 865.000000 -> 867.000000 (2.000000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: 8 x 250.000kHz
INFO  | ??:??:?? 2 Radio channel_num: 4
INFO  | ??:??:?? 2 Radio frequency: 865.875000
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=22
INFO  | ??:??:?? 2 SX126x init result 0
INFO  | ??:??:?? 2 Frequency set to 865.875000
INFO  | ??:??:?? 2 Bandwidth set to 250.000000
INFO  | ??:??:?? 2 Power output set to 22
DEBUG | ??:??:?? 2 Current limit set to 140.000000
DEBUG | ??:??:?? 2 Current limit set result 0
DEBUG | ??:??:?? 2 Setting DIO2 as RF switch
[... 85 lines cut ...]
DEBUG | ??:??:?? 47 [DeviceTelemetryModule] decoded message (id=0x6e83a016 fr=0xd3 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | ??:??:?? 47 [RadioIf] Starting low level send (id=0x6e83a016 fr=0xd3 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
DEBUG | ??:??:?? 47 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=44, time 559 ms
DEBUG | ??:??:?? 47 [RadioIf] AirTime - Packet transmitted : 559ms
DEBUG | ??:??:?? 48 [RadioIf] Completed sending (id=0x6e83a016 fr=0xd3 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
INFO  | ??:??:?? 51 [Button] Shutdown from long pressshowing shutdown screen
DEBUG | ??:??:?? 57 [GPS] NMEA GPS time 2024-02-12 22:04:21
DEBUG | ??:??:?? 57 [GPS] Upgrading time to quality 4
DEBUG | 22:04:21 57 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 22:04:21 57 [GPS] onGPSChanged() pos@0, time=1707775461, lat=0, lon=0, alt=0
INFO  | 22:04:21 57 [GPS] updatePosition LOCAL pos@0, time=1707775461, latI=0, lonI=0, alt=0
DEBUG | 22:04:21 57 [GPS] Setting local position: latitude=0, longitude=0, time=1707775461
DEBUG | 22:04:21 57 [GPS] Node status update: 1 online, 1 total
DEBUG | 22:04:24 60 [Button] Long press stop!
INFO  | 22:04:27 63 [Button] Turning off screen
INFO  | 22:04:27 63 [Button] Shutting down
INFO  | 22:04:27 63 [Button] Entering deep sleep forever
INFO  | 22:04:27 63 [Button] Disable NRF52 bluetooth
DEBUG | 22:04:27 63 [Button] SX126x entering sleep mode (FIXME, don't keep config)
INFO  | 22:04:27 63 [Button] GPS deep sleep!
DEBUG | 22:04:27 63 [Button] WANT GPS=0
DEBUG | 22:04:27 63 [Button] GPS Lock took 61, average 0
INFO  | 22:04:27 63 [Button] Setting GPS power=0
DEBUG | 22:04:27 63 [Button] Sleep Time: 120000
DEBUG | 22:04:27 63 [Button] Long press stop!

mverch67, 2024-02-12:
There seems to be an issue with your button that leads to a shutdown ... do you have the RAK in a tight enclosure that presses against the button?

vinodcombat, 2024-02-12:
I guess I found the issue. The issue is occurring with Wis Block Base board RAK19007 and GPS Module RAK1910 on Slot D of base board. Switching the gps module from Slot D to Slot A resolved the issue.

closed 2024-02-12 by thebentern