meshtastic/firmware #3219: [Bug]: Unable to configure radio if WiFi AP unavailable

todd-herbert, 2024-02-13:
### Category

WiFi

### Hardware

Heltec Wireless Paper

### Firmware Version

2.2.23.05dd680d

### Description

1. Connect to radio from Android App, using BLE
2. Configure radio to use Wifi
 2.1 *Connect to radio from Android App, providing IP address*
 2.2 *Radio working as expected*
3. External change causes WiFi AP to become unavailable (out of range, change of SSID, etc)
4. Radio continues endlessly attempting to join the unavailable AP
5. Unable to connect to radio, to reconfigure for new AP settings

___

Is there any way to exit this state, short of re-flashing the device?
There may very well be some procedure I am not aware of.

If there is not, perhaps it would be worth considering ways to handle this scenario?
A few possible options that come to mind include:
* Briefly configure for BLE at boot, before searching for WiFi
* Periodically attempt a BLE connection, after several WiFi connection failures
* Hardware keypress configures for BLE
* Hardware keypress performs factory reset

### Relevant log output

```Shell
DEBUG | 02:29:42 65 [Power] Battery: usbPower=0, isCharging=0, batMv=3916, batPct=69
[ 79019][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 79020][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
[ 79023][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
DEBUG | 02:29:55 79 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:29:55 79 WiFi lost connection. Reason: 4
INFO  | 02:29:56 80 [WifiConnect] Reconnecting to WiFi access point TestWifi
DEBUG | 02:30:02 85 [Power] Battery: usbPower=0, isCharging=0, batMv=3860, batPct=63
[ 87507][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 87507][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 87511][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 02:30:04 87 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:30:04 87 [WebServerThread] WiFi lost connection. Reason: 201
INFO  | 02:30:05 88 [WifiConnect] Reconnecting to WiFi access point TestWifi
[ 96037][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 96037][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 96041][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 02:30:12 96 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:30:12 96 WiFi lost connection. Reason: 201
INFO  | 02:30:13 97 [WifiConnect] Reconnecting to WiFi access point TestWifi
[104567][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[104567][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[104571][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 02:30:21 104 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:30:21 104 WiFi lost connection. Reason: 201
INFO  | 02:30:22 105 [WifiConnect] Reconnecting to WiFi access point TestWifi
DEBUG | 02:30:27 110 [Power] Battery: usbPower=0, isCharging=0, batMv=3916, batPct=69
[113101][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[113102][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[113105][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 02:30:29 113 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:30:29 113 WiFi lost connection. Reason: 201
INFO  | 02:30:30 114 [WifiConnect] Reconnecting to WiFi access point TestWifi
[121626][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[121627][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
[121630][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 02:30:38 121 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 02:30:38 121 WiFi lost connection. Reason: 201
INFO  | 02:30:39 122 [WifiConnect] Reconnecting to WiFi access point TestWifi
```

pdxlocations, 2024-02-13:
No need to reflash. A serial connection (USB) to an Android or web client let's you reconfigure WiFi.

todd-herbert, 2024-02-13:
Oh yes I can see that now, thank you!

closed 2024-02-13 by todd-herbert

todd-herbert, 2024-02-13:
I was completely unaware that the app could make a serial connection to the radio. The fact that the radio also powers itself from the Android device makes this a *very* nice feature!