meshtastic/firmware #3230: hostnames (device names) with underscores break RFC

cshields, 2024-02-16:
Great project, really enjoying it!

I've got a minor issue that would be a quick and easy fix, but likely disruptive to what users are used to seeing today. In short, the device names produced by `getDeviceName()` are created with an underscore between short name and the mac:

https://github.com/meshtastic/firmware/blob/e3c4bc5473a5ca9775617103aeb559e2bcc0d0ee/src/main.cpp#L154-L157

Technically, underscores are not allowed in hostnames per RFC 1035 / RFC 1123. This is not an issue within the meshtastic ecosystem that I can tell, but does become an issue with remote syslog logging when passed to parsers that match and parse messages based on those rules and constraints, causing such log entries to not be processed correctly (or not at all) depending on the destination.

The hostname used in network configs themselves are dashed (see below), but getDeviceName is used for syslog instead.

https://github.com/meshtastic/firmware/blob/e3c4bc5473a5ca9775617103aeb559e2bcc0d0ee/src/mesh/wifi/WiFiAPClient.cpp#L213-L219

Possible fixes:

1. swap out dashes where underscores are used in device names. I'm sure this would be disruptive to users who are used to the underscores today, but I would advocate for this approach as it may prevent similar issues in other use cases in the future.
2. stop using getDeviceName for the syslog hostname, possibly go with the same name advertised as the network hostname (would make sense, though I do like the shortname+mac notation)

cheers!

garthvh, 2024-02-16:
These are BLE names

cshields, 2024-02-16:
I'm less familiar with BLE naming conventions - but if they need to be underscored I'd lean toward the second fix - using network hostname in the syslog entries instead of BLE names and keep them separate?

thebentern, 2024-02-17:
@cshields would you be willing to create a PR for this fix? Either one of those fix options would need to be limited to the TCP consumers and leave the BLE naming convention to remain in it's current state.

cshields, 2024-02-21:
@thebentern sure, if you don't mind me taking a bit of time to get around to it, feel free to assign to me.

dougle03, 2024-02-28:
> @thebentern sure, if you don't mind me taking a bit of time to get around to it, feel free to assign to me.

Noe there is another issue logged on network TCP hostnames. Worth consideration in relation to this issue.