meshtastic/firmware #3235: [Feature Request]: Crypto/Channel drop when new device connected

BlackAvenger001, 2024-02-16:
### Platform

NRF52, ESP32

### Description

I'm requesting a tamper safety that drops Keys/Channels/or even does a full device wipe. 

Something like a switch in the settings that when set to 'ON' would make it so when a new/different device connects via Bluetooth or serial it just dumps Crypto or even channels all together. You could set that switch to 'OFF' via your main phone/device if you wanted to connect it to something else to do some more tweaking, but this would protect Channels and Keys from being compromised if a device is lost or stolen.

The setting could potentially have options of whether to wipe the entire device or just the channels/keys.

thebentern, 2024-02-17:
There is no way of identifying the serial host. This would have to be limited to bluetooth and possibly tcp based phone API connections

BlackAvenger001, 2024-02-17:
If there is no way of identifying serial host then the 'ON' switch in settings could initiate the drop/wipe when any device connected via serial. This would still allow your trusted person to carry the device but if it was lost or stolen then it would reset the next time someone else connected to it.

garthvh, 2024-02-17:
There is already a protobuf for this, you can build whatever you want https://github.com/meshtastic/protobufs/blob/5f28be497a5518334c86378335e8ffcd177ed661/meshtastic/admin.proto#L318

closed 2024-04-27 by garthvh