meshtastic/firmware #3240: [Feature Request]: Factory reset via hardware, such as detection sensor module or button

nagumii, 2024-02-17:
### Platform

NRF52

### Description

This would allow a hardware button or IO pin event to trigger a wiping of private data, with a focus on encryption keys and nodeDB, from storage.

One use case: a remotely placed repeater with a tamper detection mechanism that, upon detection of enclosure intrusion would wipe encryption keys, preserving the security of the mesh.

Currently, we have --factory-reset via serial. It should be easy to implement this via detection sensor, for example.

Additional functionality proposal: allow for a message to be sent to the primary channel before wiping data. Example:

1. Via some kind of sensor, enclosure intrusion (or movement of enclosure) is detected.
2. An alert bell is sent to the primary channel: "TAMPER detected, wiping memory".
3. After a timeout of say 10 seconds, firmware wipes.

GiuStel, 2024-02-27:
I agree with @nagumii ,
an hardware triggered factory reset (or personal data wipe) would be a great feature in order to protect from thefts.

garthvh, 2024-02-29:
This is already possible using the serial module it someone wants to mock up a prototype, factory reset is an admin messahe

nagumii, 2024-02-29:
Yep, I will by trying an implementation of this with a small arduino board.
Thanks

garthvh, 2024-03-10:
Looking forward to seeing the prototype, we can re-open this if there is anything needed in the firmware, but this is a perfect serial module use case!

closed 2024-03-10 by garthvh