meshtastic/firmware #3244: [Bug]: RAK18001 buzzer very quiet - solution included

tropho23, 2024-02-18:
### Category

Other

### Hardware

Rak4631

### Firmware Version

Any

### Description

I found out why the RAK18001 buzzer is so anemic and quiet! 
It's actually not...it's because we use the Arduino 'tone' command and not PWM.

This explains a few things:
1. Why it's so quiet
2. Why it gets really hot (burns your finger) when operating for more than ~1 second
3. Why it makes the RAK5005 die with the GPS module (or likely any other) installed

For #1, it's just the nature of the buzzer; it's a PWM buzzer and using the 'tone' command is not how to make it work well
For #2, same as above due to how PWM works versus constant voltage
For #3, same as #2 but the very high current draw causes a brownout condition on the RAK5005 that was resolved due to power circuit redesign for the RAK5005 and RAK19003

I have not tested this but a user on the RAK community forums has:
https://forum.rakwireless.com/t/buzzer-rak18001-weak-audio-output/10014

The submitter provided sample code to use the buzzer in PWM mode vs. 'tone' command and found the RAK18001 is plenty loud, doesn't get crazy hot while operating, and does not cause RAK5005 failure when other sensor modules are installed.

minutello, 2024-03-23:
Ok, but this doesn't explain why the current implementation using Arduino 'tone' works well with e.g. Heltec boards (I tested it).

I tried with cheap passive buzzers, mounted on the RAK IO module (not the RAK 18001 buzzer), and the problem is the same (using PWM):
* buzzer works fine with Heltec
* too quiet with RAK

I will try to solve this in hardware (maybe some resistor etc.), and report back if I got it working.

BTW, I also tried an active buzzer on the RAK IO module (of course not using PWM), and it didn't work well either. I only get an active buzzer to work with RAK if I pretend it to be a 'vibra' motor...

nagumii, 2024-04-18:
Tropho, was this solved by the PWM toggle in settings? For me the sound is very quiet, but it's not heating up.

tropho23, 2024-04-19:
PWM is required to make passive buzzers work at all, so it's not really a solution.  I suppose the best solution is to use a different buzzer, but that requires a GPIO we might not have unless a RAK13002 is added.

Tony Good

closed 2024-11-07 by github-actions[bot]