meshtastic/firmware #3249: [Bug]: T-Echo reboot loop

pista01, 2024-02-19:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.2.23.5672e68 Alpha

### Description

I have two T-Echo devices (bought shortly after they were first released) that I can't load any firmware version past 2.2.17.  I've replace the bootloader and loaded the factory erase uf2 from the firmware files.  As soon as I set the region they both start rebooting after a few minutes.

### Relevant log output

```Shell
[08:49:43.558] tio v2.7
[08:49:43.558] Press ctrl-t q to quit
[08:49:43.559] Connected
INFO  | 13:49:46 35 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | 13:49:46 35 [NodeInfoModule] sending owner !19566b76/Meshtastic 6b76/6b76
DEBUG | 13:49:46 35 [NodeInfoModule] Initial packet id 136619001, numPacketId 4294967295
DEBUG | 13:49:46 35 [NodeInfoModule] Update DB node 0x19566b76, rx_time=0
DEBUG | 13:49:46 35 [NodeInfoModule] handleReceived(LOCAL) (id=0x0824a3fb fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | 13:49:46 35 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | 13:49:46 35 [NodeInfoModule] localSend to channel 0
DEBUG | 13:49:46 35 [NodeInfoModule] Add packet record (id=0x0824a3fb fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | 13:49:46 35 [NodeInfoModule] Using AES256 key!
DEBUG | 13:49:46 35 [NodeInfoModule] Software encrypt fr=19566b76, num=824a3fb, numBytes=50!
DEBUG | 13:49:46 35 [NodeInfoModule] enqueuing for send (id=0x0824a3fb fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x7f encrypted priority=10)
DEBUG | 13:49:46 35 [NodeInfoModule] txGood=0,rxGood=0,rxBad=0
DEBUG | 13:49:46 35 [RadioIf] Starting low level send (id=0x0824a3fb fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x7f encrypted priority=10)
DEBUG | 13:49:46 35 [RadioIf] (bw=125, sf=11, cr=4/8) packet symLen=16 ms, payloadSize=66, time 2396 ms
DEBUG | 13:49:46 35 [RadioIf] AirTime - Packet transmitted : 2396ms
DEBUG | 13:49:46 36 [Button] Updating E-Paper... done
DEBUG | 13:49:58 48 [Power] Battery: usbPower=1, isCharging=1, batMv=4631, batPct=100
INFO  | 13:50:01 50 [DeviceTelemetryModule] (Sending): air_util_tx=0.066556, channel_utilization=3.993334, battery_level=101, voltage=4.631000
DEBUG | 13:50:01 50 [DeviceTelemetryModule] updateTelemetry LOCAL
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Node status update: 0 online, 1 total
DEBUG | 13:50:01 50 [DeviceTelemetryModule] showing standard frames
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Showing 0 module frames
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Total frame count: 103
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Added modules.  numframes: 0
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Finished building frames. numframes: 2
INFO  | 13:50:01 50 [DeviceTelemetryModule] Sending packet to mesh
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Update DB node 0x19566b76, rx_time=0
DEBUG | 13:50:01 50 [DeviceTelemetryModule] handleReceived(LOCAL) (id=0x0824a3fc fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | 13:50:01 50 [DeviceTelemetryModule] No modules interested in portnum=67, src=LOCAL
DEBUG | 13:50:01 50 [DeviceTelemetryModule] localSend to channel 0
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Add packet record (id=0x0824a3fc fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Using AES256 key!
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Software encrypt fr=19566b76, num=824a3fc, numBytes=28!
DEBUG | 13:50:01 50 [DeviceTelemetryModule] enqueuing for send (id=0x0824a3fc fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x7f encrypted priority=1)
DEBUG | 13:50:01 50 [DeviceTelemetryModule] txGood=0,rxGood=0,rxBad=0
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Using channel 0 (hash 0x7f)
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Using AES256 key!
DEBUG | 13:50:01 50 [DeviceTelemetryModule] Software encrypt fr=19566b76, num=824a3fc, numBytes=28!
DEBUG | 13:50:01 50 [DeviceTelemetryModule] decoded message (id=0x0824a3fc fr=0x76 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | 13:50:01 50 [Screen] Updating E-Paper... done
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
WARN  | 13:50:02 51 [RadioIf] Can not send yet, busyTx
[... 948 lines cut ...]
DEBUG | 13:50:50 99 [Screen] Screen: Critical error 8!
WARN  | 13:50:50 99 [RadioIf] Can not send yet, busyTx
ERROR | 13:50:50 99 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 13:50:50 99 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:88
DEBUG | 13:50:50 99 [Screen] Screen: Critical error 8!
WARN  | 13:50:50 100 [RadioIf] Can not send yet, busyTx
ERROR | 13:50:50 100 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 13:50:50 100 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:88
DEBUG | 13:50:50 100 [Screen] Screen: Critical error 8!
WARN  | 13:50:50 100 [RadioIf] Can not send yet, busyTx
ERROR | 13:50:50 100 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 13:50:50 100 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:88
DEBUG | 13:50:50 100 [Screen] Screen: Critical error 8!
WARN  | 13:50:51 100 [RadioIf] Can not send yet, busyTx
ERROR | 13:50:51 100 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 13:50:51 100 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:88
DEBUG | 13:50:51 100 [Screen] Screen: Critical error 8!
WARN  | 13:50:51 100 [RadioIf] Can not send yet, busyTx
ERROR | 13:50:51 100 [RadioIf] Hardware Failure! busyTx for more than 60s
ERROR | 13:50:51 100 [RadioIf] NOTE! Recording critical error 8 at src/mesh/RadioLibInterface.cpp:88
DEBUG | 13:50:51 100 [Screen] Screen: Critical error 8!

[08:50:51.442] Disconnected
[08:50:52.444] Connected
```

pista01, 2024-02-21:
UF2 Bootloader 0.6.1-2-g1224915 lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90f) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: LilyGo T-Echo
Board-ID: nRF52840-TEcho-v1
SoftDevice: S140 version 6.1.1
Date: Feb 10 2024

pista01, 2024-02-27:
Same issue with 2.2.24

thebentern, 2024-02-27:
I'm unable to reproduce this on multiple T-Echos. Does rolling back to a previous release resolve the issue?

pista01, 2024-02-27:
Yes, I just rolled it back to 2.2.17 and it's working fine.  I have 2 T-Echos like this.  Anything above 2.2.17 doesn't work.

pista01, 2024-02-27:
I'd be happy to ship one of these T-Echos

thebentern, 2024-02-28:
@pista01 is this link accessible? If so, can you try the uf2 in the zip?
https://github.com/meshtastic/firmware/actions/runs/8071539015/artifacts/1280390603
I created a revert PR based on a suspicion given .17 was your last working version.

pista01, 2024-02-28:
Looks like that fixed it.  Both devices are connected to the 8 other nodes with no reboots.  Thank you!

andrekir, 2024-02-29:
@pista01 out of curiosity, what color is the activity led on your T-Echo after flashing the above .uf2 file with the fix?

pista01, 2024-02-29:
Blue

thebentern, 2024-03-03:
Fixed with #3304

closed 2024-03-03 by thebentern