meshtastic/firmware #3254: [Feature Request]: Allow bluetooth connection when network is not connected

drewzh, 2024-02-20:
### Platform

ESP32

### Description

### Problem Summary
I have a node that I like to have connected via WiFi... but on occasion, I'd like to be able to connect via Bluetooth, but can't because the node is 'locked' in WiFi/Ethernet mode. When I'm out of the house, this essentially means I can't use node and I have to wait until I return to change the mode over to bluetooth.

### Suggested Solution
Instead of completely disabling Bluetooth when Eetwork is enabled, Ideally, if the network connection is not available, I'd still like to be able to connect via Bluetooth until the network connection is established - this would allow remote admin in the event that the network is unavailable.

If a connection is made via bluetooth, the network connection routine should pause until a later time when bluetooth is not currently being used.

thebentern, 2024-02-20:
This is impossible due to a hardware limitation. Wifi and Bluetooth use the same radio on the ESP32

closed 2024-02-20 by thebentern

garthvh, 2024-02-20:
They use the same antenna, not possible

drewzh, 2024-02-20:
> This is impossible due to a hardware limitation. Wifi and Bluetooth use the same radio on the ESP32

Ah - I knew they used the same radio, but I was not aware that they couldn't be operated independently from one another.

That said, with the problem statement still holding for me and I guess others - is there not another way to tackle this problem...

A suggestion might be, if the device has just powered on and a network connection hasn't been made in X tries or X seconds, then fallback to Bluetooth? Or perhaps always enable bluetooth first for 60 seconds and then keep the bluetooth connection alive until the end of the connected session and then fallback to network again? Just *something* to allow someone to connect to the node if the network fails?

Perhaps even add a long-press handler on one of the status screens to toggle network on and off for devices such as the Heltec V3 that has buttons?

I wasn't aware of the radio limitations, but that doesn't mean an alternative solution cannot be found, surely?

drewzh, 2024-02-20:
> Some friendly advice that would help both you and those who maintain this repository: anytime you have an idea, I would suggest doing a quick search in closed issues before opening a new one to see if it has already been addressed.

I searched but didn't come up with anything relevant to my problem on the first check. That said - I can see there's something related here after changing my search terms, but the same problem has yet to be discussed with a real solution proposed. https://github.com/meshtastic/firmware/pull/2180

We don't need both bluetooth and network to be enabled at the same time to allow a mechanism to get out of this 'network-enabled lockout' situation.

The node becomes a paperweight from an interactivity standpoint until a compatible router can be found to connect via or a serial connection is made... neither of these things are available when mobile, nor would they likely be available in a situation where WiFi suddenly goes offline in an emergency.

drewzh, 2024-02-20:
> This is impossible due to a hardware limitation. Wifi and Bluetooth use the same radio on the ESP32

As per my other comments RE this not being a hardware limitation but rather a usability issue (a network misconfiguration shouldn't lock a user out from the device) - could you possibly re-open it? If not, where else can we discuss potential solutions to this issue? Thanks @thebentern @rcarteraz @garthvh

tropho23, 2024-02-20:
> As per my other comments RE this not being a hardware limitation but rather a usability issue (a network misconfiguration shouldn't lock a user out from the device) - could you possibly re-open it? [...]

I had the same issue in the past: I mistyped the IP address, and now device cannot be accessed to fix it; but actually it _can_ be accessed, via USB cable. In my case I simply re-flashed firmware and went on my way.

Yes, I am sometimes too careless to carefully type but we cannot have safeguards for every single edge case.  We must trust the user to make deliberate decisions.  Reflash firmware and retry, works every time.

jp-bennett, 2024-02-20:
IMO, it's reasonable to think about a way to disable WiFi and re-enable bluetooth when the WiFi network is unavailable.

tropho23, 2024-02-20:
> IMO, it's reasonable to think about a way to disable WiFi and re-enable bluetooth when the WiFi network is unavailable.

Four presses of the user button to disable WiFi?  We risk confusing people as we already sometimes do with triple-press disabling GPS.  I really like that feature but it has the potential for accidental GPS disable.

drewzh, 2024-02-20:
I use hardwired (PoE), WiFi, and Bluetooth nodes and my desire and understanding for my network nodes to stay connected no matter what and not have Bluetooth or accidental button presses (which are far less likely with base stations), not get in the way of that is clear... I'm all for that. That said, there is a genuine need for the network option to be toggled easily and without the need of a computer if for e.g, the network goes away in an emergy power outage situation or if I suddenly have to repurpose a node in an emergency (or hey, just the convenience of being able to grab and node and quickly change it's usage from base to mobile - which is what always catches me out).

Would a button press on power-up be capturable by the meshtastic firmware? I understand this wouldn't capture all devices however as some might not have an easily pressable button, but as I understand it, this "no bluetooth when network connected" only affects ESP32 devices, which all if not most have a button available. I know my RAK4631 happily accepts bluetooth connections when connected via the network.

thebentern, 2024-02-20:
> I was thinking more about a fallback option, where no WiFi connection within the first minute would re-enable bluetooth. Maybe even switch back and forth every 60 seconds until one connects.

Wouldn't this just generate support requests where people are confused as to why their wireless network connection inevitably disconnects and switches to bluetooth instead of just disconnecting?

drewzh, 2024-02-20:
> > Four presses of the user button to disable WiFi?
> 
> I was thinking more about a fallback option, where no WiFi connection within the first minute would re-enable bluetooth. Maybe even switch back and forth every 60 seconds until one connects.

I also thought about the cycling idea as this would allow both mechanisms to continue working... it seems a little janky as a software dev, but it's entirely functional and could be hidden behind a new toggle in the network tab "Attempt to cycle between BLE/WiFi until connection established".

The button press on startup idea seems a cleaner, more explicit approach if that can be handled. You could even have this option be configurable as a "Long press X button on startup behavior: [factory reset] / [toggle network] / [clear nodedb]".

drewzh, 2024-02-20:
> Wouldn't this just generate support requests where people are confused as to why their wireless network connection inevitably disconnects and switches to bluetooth instead of just disconnecting?

Doesn't have to be default - just an option in the network tab to enable the behavior. That said, I'd prefer other options for the reasons mentioned.

On the topic of having the firmware handle this - what about having bluetooth come up for 60 seconds first before attempting to connect to network when no BLE connection is made? If a BLE connection is made, keep the device in that mode until reboot?

jp-bennett, 2024-02-20:
> why their wireless network connection inevitably disconnects and switches to bluetooth instead of just disconnecting

After one or the other connects, we'd probably leave that enabled for the rest of runtime. Just cycle between the two on first boot.

roboraptor, 2025-05-23:
Integrate a new switch into the Wireless settings UI. Call it "Traveller Mode" or something similar.
Users who frequently take their device outside will find it most useful.
Once I'm home and my wifi is detected, connect to it. (probably when I press reset)
If the wifi disconnects and is unavailable for a period, it switches to bluetooth. 
I find nothing confusing about that.

I like the Bluetooth concept too. :)