meshtastic/firmware #3257: [Bug]: Critical Error 3 on heltec lora 32 V1 SX1276

AndyKirby, 2024-02-21:
### Category

Hardware Compatibility

### Hardware

Heltec v1

### Firmware Version

v1-2.2.19.8f6a283, 1-2.2.22.404d0dd, v1-2.2.23.5672e68

### Description

Firmwares from repository load OK and pairing goes as usual but after setting the lora to EU868 the screen shows critical error 3.

Tried all three firmwares above bu none work. I have a heltec 433 lora 32 V1 (sx1278) and this works fine configured as EU433.

Problem seems to be with the SX1276 radio chips and provided firmwares.

### Relevant log output

```Shell
WARN  | ??:??:?? 0 SPI.begin(SCK=5, MISO=19, MOSI=27, NSS=18)
DEBUG | ??:??:?? 0 Read RTC time as 0
INFO  | ??:??:?? 0 Setting GPS power=1
DEBUG | ??:??:?? 0 WANT GPS=1
INFO  | ??:??:?? 0 Setting GPS power=1
DEBUG | ??:??:?? 0 Using GPIO36 for GPS RX
DEBUG | ??:??:?? 0 Using GPIO33 for GPS TX
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
INFO  | ??:??:?? 0 External Notification Module Disabled
INFO  | ??:??:?? 0 Turning on screen
DEBUG | ??:??:?? 0 Module wants a UI Frame
WARN  | ??:??:?? 0 RF95Interface(cs=18, irq=26, rst=14, busy=-1)
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 0 Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 1
INFO  | ??:??:?? 0 Radio frequency: 869.525024
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Set radio: final power level=20
INFO  | ??:??:?? 0 RF95 init result -16
INFO  | ??:??:?? 0 Frequency set to 869.525024
INFO  | ??:??:?? 0 Bandwidth set to 250.000000
INFO  | ??:??:?? 0 Power output set to 20
WARN  | ??:??:?? 0 Failed to find RF95 radio
INFO  | ??:??:?? 0 Not using WIFI
ERROR | ??:??:?? 0 NOTE! Recording critical error 3 at src/main.cpp:916
```

closed 2024-02-21 by thebentern

AndyKirby, 2024-02-21:
Then stop building firmwares and releasing them.

And I had already checked the error code before posting the bug.

thebentern, 2024-02-21:
> If you are sending mixed messages expect people to file bug reports when the issued firmwares fail.

Fair point. I'm removing it from the CI/CD assets