meshtastic/firmware #3260: [Bug]: Firmware reset after battery dies

DomTheDorito, 2024-02-22:
### Category

Other

### Hardware

Heltec v2

### Firmware Version

2.2.21

### Description

Device settings get reset after battery dies (region, channels, users etc). 

To reproduce issue:

1. While device is powered on, allow battery to run flat
2. Plug in the charge battery/power device
3. Region becomes unset, and acts like its a new device, asks to re-pair to phone

pdxlocations, 2024-02-22:
I'm curious, Is the node connected to a client by Bluetooth when the battery dies? Does it reset when NOT connected via Bluetooth?

DomTheDorito, 2024-02-22:
It's done it in both cases now.

pdxlocations, 2024-02-22:
Ok thanks, I had a brownout reset issue with nRF and others have reported resets when losing Bluetooth connection so I was curious if they might me related. #2936

DomTheDorito, 2024-02-22:
I'll do some more testing tomorrow afternoon, and I'll try and get some useful logs

Hamspiced, 2024-02-26:
Same issue, rak 4631 with a 1000mah flat pack.

When mine resets though the ble pin switches to random and I have to reset via serial

cracky22, 2024-03-03:
same issue with my heltec v2.1....

OrByR, 2024-03-07:
I have run into this issue at least once on a Heltec v3 as well.

Holyszewski, 2024-04-13:
In my case, on a Heltec v3 set up as a router, the brownout just resets the long and short name. The other settings seem to be preserved and it allows me to fix the name settings remotely via the admin channel.

FW 2.3.2 I believe.

cracky22, 2024-04-13:
Yep and finally my heltec v3 now also has these problems

Hamspiced, 2024-04-16:
> Does this happen every time the battery dies, or just sometimes?

It is infrequent.  Sometimes ill go 2-3 charges from dead before it pops up.  Sometimes a week.

The Heltec Wireless Stick Lite seems to have it happen much more frequently than my other devices.

Recently a solar node/rak wireless build had this occur.  Normally not an issue but it was mounted 30' up.  Definately a pita to get it down and set it up again.  This particular device was setup as a Router/Client  the other devices i had it occur on were only clients.

cracky22, 2024-04-16:
> Does this happen every time the battery dies, or just sometimes?

For me the problem only occurs very sporadically. sometimes it can take weeks to occur but sometimes three to four days are enough...

WUBBSY, 2024-05-19:
> In my case, on a Heltec v3 set up as a router, the brownout just resets the long and short name. The other settings seem to be preserved and it allows me to fix the name settings remotely via the admin channel.
> 
> 
> 
> FW 2.3.2 I believe. 

Had the same issue on both my Heltec Tracker nodes after the battery went flat. 
Set up as Clients. 
FW 2.3.3

fattredd, 2024-05-29:
I'm also seeing the same on 2.3.10 on Heltec V3. Set up as Router Clients.

cliffjao, 2024-09-08:
I'm running into this issue on rak4631 but battery is still solid at 95%.

tedwardd, 2024-09-11:
Still seems to be an issue with the Heltec V3 using 2.5.0.e470619 preview. All settings lost after battery depletion. Perhaps a "safe shutdown" feature would be a reasonable stop-gap solution here. Similar to power off on power loss but "power off at X percentage battery remaining"?

droptic, 2024-09-14:
Same problem in lora32 v2.1_1.6

dannypagano, 2024-09-23:
Same issue with my Heltev V3 on 2.4.2. All settings lost after battery runs flat. Wants to re-pair, region is reset, heartbeat LED setting resets, channels lost, etc. On iOS, the Meshtastic app still sees the device via bluetooth, but fails to automatically reconnect, and requires you to forget the bluetooth device and re-pair. 

It's happened every single time the battery runs flat (~10-12 hours 🫠). Always connected to the phone via bluetooth when the battery dies.

I have another identical device that I'll test to see if the same issue persists.

CyberRatingsTim, 2024-10-02:
Same on Heltec v3 based on the ESP32 i think. 
Every time the battery dies all the way, it loses settings and acts like a new device to pair with.

CyberRatingsTim, 2024-10-02:
before 2.49V it goes into a boot loop. and loses the settings

cracky22, 2024-10-24:
I’m now at the point where the problem occurs daily on my Heltec V3 and V2 (I know that the V2 is no longer supported) when the battery drops below ~40%. This is quite frustrating! As a V2 user, is there still a way for me to fix this issue on my V2 board once the bug is resolved, considering that the update files are no longer provided via GitHub release files?

adingbatponder, 2024-10-29:
https://github.com/orgs/meshtastic/discussions/5#discussion-7387496

fifieldt, 2024-11-16:
Did anyone see an improvement after using 2.5.10 or 2.5.11 ? There were some filesystem related changes.

cracky22, 2024-11-16:
> Did anyone see an improvement after using 2.5.10 or 2.5.11 ? There were some filesystem related changes.

Hi, I haven't seen anything spontaneously yet but I can only run my devices with battery and observe

cracky22, 2024-11-17:
> Did anyone see an improvement after using 2.5.10 or 2.5.11 ? There were some filesystem related changes.

Good morning, the bug seems to have been fixed on my devices. I've drained them twice and all the data is still there. Great work!
I have another question: Can I now fix THIS bug on the no longer supported Heltec v2.1 models ?

fifieldt, 2024-11-17:
Drop by #firmware on discord and I'm sure someone will be happy to build a firmware for you :)

fifieldt, 2024-11-17:
Closing this one since it appears to be fixed. We're still working on the related one here: https://github.com/meshtastic/firmware/issues/4555

bittylicious, 2026-03-14:
Like @brunoarine I've just noticed this with my Heltec v4. It was connected to a 3000mah battery which died, and I lost all settings.