meshtastic/firmware #3262: [Bug]:

SignalMedic, 2024-02-22:
### Category

Other

### Hardware

Other

### Firmware Version

2.2.23.5672e68 Alpha

### Description

When using with NRF and ESP boards and the latest ATAK plugin I am not able to achieve functionality within ATAK.

I have tried with ATAK 4.10 and ATAK 5.0 both CIV and MIL for each.

The green/red iMesh connection status indicator behaves as expected.

No contacts, pli, COT, chats, or otherwise appear to work using the Meshtastic network.

I have changed the Mesthastic device to TAK. I have also tried other device settings. Of note, changing the device type does not appear to change any ATAK behavior or connection status..

thebentern, 2024-02-27:
There is not enough information to diagnose what is going on. We need serial logs from the device.

thebentern, 2024-02-29:
This has likely been resolved in the latest ATAK plugin release
https://github.com/meshtastic/ATAK-Plugin/releases/tag/1.0.6