meshtastic/firmware #3266: [Feature Request]: Allow longer short name or add dedicated callsign / town field.

robbietee, 2024-02-22:
### Platform

NRF52, ESP32

### Description

Allow longer short name so that more useful names can be used. For example Amateur Radio Callsign or Node Town / Postcode
Or create a seperate dedicated Callsign / Town field.

Many thanks to the Developers. Keep up the good work.

garthvh, 2024-02-23:
Long Name is what is used for call sign

closed 2024-02-23 by garthvh

robbietee, 2024-02-23:
I know long name is currently used. That's why I have asked if short name can be made longer so that can be used instead. Short name isn't actually very practical being just 4 characters is my point. Callsign was just one example of a better use.

garthvh, 2024-02-23:
Adding more data to the packet increasing the airtime is just going to make these problems worse.  The short name is used extensively in the apps because of it's constrained size.