meshtastic/firmware #3270: [Bug]: wifi disconnects constant

Matzebhv, 2024-02-23:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.2.22.404d0dd

### Description

Wifi is very unstable and disconnects constantly. You can force the disconnect hitting refresh in your browser. Device reconnects after ca. 30 seconds, Standard settings as router_client eu_868 mqtt enabled.

GUVWAF, 2024-02-24:
See https://github.com/meshtastic/firmware/discussions/3031 and the note [here](https://meshtastic.org/docs/configuration/module/mqtt/#connect-to-the-default-public-server). There is a lot of traffic on the default channel on the public MQTT server (not only text, but all kind of messages), which the device simply cannot handle. 
You could try only using MQTT on a different channel than the default.

Matzebhv, 2024-02-24:
Have disabled downlink on primary for now and will look kat it. But even with downlink disabled you can trigger this behavior using your browser to connect to the device.

closed 2024-02-27 by thebentern