meshtastic/firmware #3273: [Bug]: Position not sent

todd-herbert, 2024-02-24:
### Category

Other

### Hardware

Heltec Wireless Paper

### Firmware Version

2.2.24.b2ea1e23

### Description

1. Erase flash
2. Build and upload for Heltec Wireless Paper
3. Disable smart position, decrease GPS broadcast interval to 10 seconds (for testing)
4. Use fixed position
    Other nodes do not receive position, even manually requested from android app.
    Log indicates that position is not being send because precision is 0.
    Android app has no field for precision.

5. Connect via serial, at client.meshtastic.org
Channel Precision field, in position settings, is blank

6. Set channel precision. Tested values 1 and 32
7. Save settings
  device reboots
8. Open positions settings
    Channel precision field is again blank

### Relevant log output

```Shell
INFO  | 06:28:04 70 [PositionModule] Sending pos@0:6 to mesh (wantReplies=0)
DEBUG | 06:28:04 70 [PositionModule] Skipping location send because precision is set to 0!
DEBUG | 06:28:04 70 [PositionModule] allocReply returned a nullptr
```

todd-herbert, 2024-02-24:
Thanks for taking the time to update. Sorry if it's just a case of jumping the gun, just wasn't sure if it needed to be brought to anyone's attention. It sounds like it's already well under control.

jp-bennett, 2024-02-24:
I'm pretty sure the old default behavior will work again with PR #3278

closed 2024-02-24 by jp-bennett