meshtastic/firmware #3284: [Feature Request]: Add UIFrame for PaxCounter module

thoherr, 2024-02-25:
### Platform

ESP32

### Description

Currently the PaxCounter module added by @caveman99 only sends the gathered information via the mesh.
A display of the current counter (WiFi/BLE/uptime) via an UI frame for the module would be useful.

I already implemented a prototype for this by adding the implementation of method drawFrame() and enabling wantUIFrame(), tested on my LilyGo LoRa32 V2.1_1.6.

caveman99, 2024-02-25:
is this a local display (the sending node) or remote, like with the environment telemetry?

Either way, please open up a PR, i'd gladly review.

thoherr, 2024-02-25:
> is this a local display (the sending node) or remote, like with the environment telemetry?

currently only local display, i'm just getting into this stuff.
a remote display would certainly make sense as well, sould be the next step.

i'll open a PR for my changes, thanks for considering it.

closed 2024-02-27 by thebentern