meshtastic/firmware #3289: [Bug]: Waveshare rp2040 OLED screen critical error 3

2E0WKS, 2024-02-25:
### Category

Other

### Hardware

Raspberry Pi Pico (W)

### Firmware Version

2.2.20. 404dodd

### Description

Waveshare rp2040 If pin 3 and pin 4 are used for oled screen it causes screen to work and LoRa radio unavailable.
Moving oled to pin 6 and results in oled screen not being found, however LoRa radio is operational
Have tried compiling firmware with pins and it seems that they are not scan I2C on boot.
Secondly waveshare 22dbm and defaults to 27dbm

GUVWAF, 2024-02-26:
You're referring to the [RP2040-LoRa](https://www.waveshare.com/rp2040-lora.htm) ("rp2040-lora" in the .uf2 file), and not the Waveshare SX1262 hat for the Pico, correct? Can you show the output of the serial logs using the [CLI](https://meshtastic.org/docs/software/python/cli/installation/) command `meshtastic --noproto` or any serial monitor on baud 115200 when you connected the screen when the LoRa radio is unavailable?

The power setting depends on the region you set (27dBm is `EU_868` for example), but the actual value that is used will be enforced by the radio library.

2E0WKS, 2024-02-27:
Hi 2.2.23 firmware caused same issue, however I saw a banner for sx1262 issues.
Today I used platformio.io to build 2.2.24 with my rp2040 LoRa and moved display to pin 4 and pin 5 and it works with display and no critical error 3.
Thank you for taking the time to come back to me, the later compiled firmware resolves the issues.
Whatever had broken has resolved itself I believe

closed 2024-02-27 by 2E0WKS