meshtastic/firmware #3294: [Bug]: Heltec Tracker is broken with new changes

thebentern, 2024-02-26:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

Current master

### Description

@Gabrielerusso something is not right with the Heltec Wireless Tracker setup now. I'm getting false positives on i2c device scan and all of kinds of other strange regressive behaviors

### Relevant log output

```Shell
DEBUG | ??:??:?? 1 Using analog input 1 for battery level
INFO  | ??:??:?? 1 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 1 I2C device found at address 0x1
INFO  | ??:??:?? 1 Device found at address 0x1 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x2
INFO  | ??:??:?? 1 Device found at address 0x2 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x3
INFO  | ??:??:?? 1 Device found at address 0x3 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x4
INFO  | ??:??:?? 1 Device found at address 0x4 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x5
INFO  | ??:??:?? 1 Device found at address 0x5 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x6
INFO  | ??:??:?? 1 Device found at address 0x6 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x7
INFO  | ??:??:?? 1 Device found at address 0x7 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x8
INFO  | ??:??:?? 1 Device found at address 0x8 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x9
INFO  | ??:??:?? 1 Device found at address 0x9 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xa
INFO  | ??:??:?? 1 Device found at address 0xa was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xb
INFO  | ??:??:?? 1 Device found at address 0xb was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xc
INFO  | ??:??:?? 1 Device found at address 0xc was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xd
INFO  | ??:??:?? 1 Device found at address 0xd was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xe
INFO  | ??:??:?? 1 Device found at address 0xe was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0xf
INFO  | ??:??:?? 1 Device found at address 0xf was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x10
INFO  | ??:??:?? 1 Device found at address 0x10 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x11
INFO  | ??:??:?? 1 Device found at address 0x11 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x12
INFO  | ??:??:?? 1 PMSA0031 air quality sensor found
DEBUG | ??:??:?? 1 I2C device found at address 0x13
INFO  | ??:??:?? 1 Device found at address 0x13 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x14
INFO  | ??:??:?? 1 Device found at address 0x14 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x15
INFO  | ??:??:?? 1 Device found at address 0x15 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x16
INFO  | ??:??:?? 1 Device found at address 0x16 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x17
INFO  | ??:??:?? 1 Device found at address 0x17 was not able to be enumerated
DEBUG | ??:??:?? 1 I2C device found at address 0x18
[... 65 lines cut ...]
INFO  | ??:??:?? 7 Device found at address 0x38 was not able to be enumerated
DEBUG | ??:??:?? 7 I2C device found at address 0x39
INFO  | ??:??:?? 7 Device found at address 0x39 was not able to be enumerated
DEBUG | ??:??:?? 7 I2C device found at address 0x3a
INFO  | ??:??:?? 7 Device found at address 0x3a was not able to be enumerated
DEBUG | ??:??:?? 7 I2C device found at address 0x3b
INFO  | ??:??:?? 7 Device found at address 0x3b was not able to be enumerated
DEBUG | ??:??:?? 7 I2C device found at address 0x3c
INFO  | ??:??:?? 7 sh1106 display found
DEBUG | ??:??:?? 7 0x0 subtype probed in 1 tries
DEBUG | ??:??:?? 8 I2C device found at address 0x3e
INFO  | ??:??:?? 8 Device found at address 0x3e was not able to be enumerated
DEBUG | ??:??:?? 8 I2C device found at address 0x3f
INFO  | ??:??:?? 8 st7567 display found
DEBUG | ??:??:?? 8 I2C device found at address 0x40
DEBUG | ??:??:?? 8 Wire.available() = 0
DEBUG | ??:??:?? 8 Register MFG_UID: 0x0
INFO  | ??:??:?? 8 INA219 sensor found at address 0x40
DEBUG | ??:??:?? 9 I2C device found at address 0x42
INFO  | ??:??:?? 9 INA3221 sensor found at address 0x42
DEBUG | ??:??:?? 9 I2C device found at address 0x43
INFO  | ??:??:?? 9 Device found at address 0x43 was not able to be enumerated
DEBUG | ??:??:?? 9 I2C device found at address 0x44
INFO  | ??:??:?? 9 SHT31 sensor found
```

Gabrielerusso, 2024-02-26:
the only edit of i2c was to actually define the i2c pins, before they were not defined at all, now they are defined as 
```
SDA = 45
SCL = 46
```
I'm waiting for some compatible i2c sensors to arrive to do some testing, but those "errors" looks no-sense to me.

Gabrielerusso, 2024-02-26:
could that be the case that as the HW has no pull-ups on those pin, this is causing false positives?

Gabrielerusso, 2024-02-26:
> According to the Heltec provided schematics, for the V.03 (oldest) GPIO 45 & 46 are LED K and VTFT Ctrl, while on the V.05 (newest) Those GPIO's are just brought out to P5 with no pullups. A quick test would be to add some 1K resistors to pins 1 & 2 to 3.3V.

Yes, my bad, i simply copy-pasted the setup for V1.1 (actual V0.5 from datasheet), for older 1.0 board other pin other than 45/46 should be used, like 5 and 6 .

GPSFan, 2024-02-26:
Of course, now i can't upload to the WT, keep getting the following error, tried holding the user button, tried holding it and then a reset...

CURRENT: upload_protocol = esp-builtin
Uploading .pio/build/heltec-wireless-tracker/firmware.bin
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: could not find or open device!
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

GPSFan, 2024-02-26:
will do

GPSFan, 2024-02-26:
finally got it flashed. I'm not seeing the above i2c oddities without the pullups.
I did a factory reset to be sure and still no oddities.

DEBUG | ??:??:?? 1 Using analog input 1 for battery level
INFO  | ??:??:?? 1 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
INFO  | ??:??:?? 1 No I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=48, swver=2.2.24.48ebf937

thebentern, 2024-02-26:
Occurring for me specifically on the v1.

Gabrielerusso, 2024-02-26:
> Occurring for me specifically on the v1.

V1.?

thebentern, 2024-02-27:
V1.0

Gabrielerusso, 2024-02-27:
> V1.0

Yes, as said before I used the wrong pin number for 1.0, 45/46 are already used,  5/6 may be better.
This one should fix it.
https://github.com/Gabrielerusso/firmware/tree/heltec-tracker-1.0-bugfix

thebentern, 2024-02-27:
I reverted the i2c changes before I saw your comment. Works as expected now. We'll try your recommended pins 5 and 6 and see how it behaves.

> > V1.0
> 
> Yes, as said before I used the wrong pin number for 1.0, 45/46 are already used, 5/6 may be better. This one should fix it. https://github.com/Gabrielerusso/firmware/tree/heltec-tracker-1.0-bugfix
> 
> [Gabrielerusso@652e5c4](https://github.com/Gabrielerusso/firmware/commit/652e5c47ce10fc1fa607e0ddb7f71e18f2068622)

closed 2024-02-27 by thebentern