meshtastic/firmware #3296: [Feature Request]: Second MQTT Client for API & Statistic use (to integrate into home assistant)

Lexorius, 2024-02-26:
### Platform

NRF52, ESP32

### Description

Hello There,

i have to suggestion to create a 2nd mqtt client for Statistic and API usages. 

what would i love to have in the api :
- statistics. how is the status of the gateway / airquality etc
- last message(es) 
- posibility to send a message to a specific user .

why a 2nd  mqtt ? 

because the first is used for messaging. 

Greetings from Germany 

Thomas

thebentern, 2024-02-27:
This is too resource intensive for the devices to handle. Embedded MQTT and the TCP api is already a heavy lift for even the ESP32.

closed 2024-02-27 by thebentern

Lexorius, 2024-02-28:
hi,
sad. thank you for your answer. ;) 

greetings from Germany and Thank you :) 

Thomas

Slyke, 2024-03-17:
@thebentern would it be better to have some Docker container running on a PC that's grabbing all events from the Meshtastic node and have that used to fulfill the above feature? Could be done over say a serial connection?

lucidnx, 2024-04-01:
I would prefer over TCP... just some docker container that will receive messages and you will decide what to do with them.