meshtastic/firmware #3300: [Feature Request]: Discern neighbor nodes from the others in nodes list

GiuStel, 2024-02-27:
### Platform

NRF52, ESP32

### Description

One of the recurrent questions asked by newbies is, in my opinion, how to know what are the "zero-hops" nodes.
Neighbor module and results analysis is the only way to answer that question at the moment, but this is not easy to set-up nor immediate.

I would suggest something that highlights the zero-hops nodes in the nodes list.

Thanks

thebentern, 2024-02-27:
This is a duplicate (albeit with a different mechanism than neighbors module) in terms of functionality. 
#3239
https://github.com/meshtastic/rfcs/pull/8

closed 2024-02-27 by thebentern

roha-github, 2024-05-10:
How does the neighbor module in the Android app show me my direct neighbors who receive and forward my messages? 

I would also like to see in the app which node I am using to receive messages from the Mesh.

Background: I have 50 nodes in the list here, but they are all more than 15km away. Until now, I have always had to use traceroute to manually determine via which nodes I am communicating.

garthvh, 2024-05-10:
Neighbors are those nodes without a hops away number, this data will improve over time as people upgrade their firmware.

roha-github, 2024-05-11:
I think there is now need to enable or update others firmware to collect my neighbors. My firmware could collect all nodes, that re-transmit data from my node. Or I'm wrong?

I have enabled neighbors module but found no visualization inside the Android app. The MQTT map https://meshtastic.liamcottle.net has sometimes neighbor informations.