meshtastic/firmware #3303: Traceroute not indicating MQTT in route

Danie10, 2024-02-27:
We're trying to get connection between some nodes at extreme distance, so have been using traceroute on the mobile app to try and see whether we're getting RF to RF, or whether it is going via MQTT between us.

Problem is, the traceroute always shows radio name to radio name. I was expecting that if there was no RF link, it would indicate radio name --> MQTT --> radio name?

This is also so for two nodes that have never had any RF link yet (also no nodes between to even establish a link) so they have only communicated via MQTT. Traceroute indicates direct link, and even the list of peers shows the node with "snr".

Image below shows example of traceroute between two nodes communicating over MQTT:

[image]

Danie10, 2024-02-28:
That would be perfect .

GUVWAF, 2024-02-28:
Recently a `via_mqtt` tag was already added to the MeshPacket:
https://github.com/meshtastic/firmware/blob/6acc63729b702501dba4a83e7285f068d6a21d13/src/mqtt/MQTT.cpp#L147

So I think this can go back to the Android repo :)

Danie10, 2024-03-02:
@mc-hamster should this then move back to Android seeing the above change was made to firmware?