meshtastic/firmware #3311: [Feature request]: MQTT Server Config per Channel

erayymz, 2024-03-01:
Currently, I can configure an mqtt server in the settings and I can decide to enable which channels to use mqtt server that I configured. 
This limits the usage of only one mqtt server per node.

As a user, I want to be able to add more than one mqtt server in the settings then choose which mqtt server I want to use for each channel so that I can use a public mqtt for say long_fast and private mqtt server for other channels that I create.

Acceptance Criteria:
Navigate to Settings > MQTT. Turn on the slider for MQTT enabled. I add more than one mqtt server and save them. Perhaps ability to add a descriptive name for each server would be helpful. Then I navigate to Settings > Channels > One of the channels. In addition to Uplink Enabled and Downlink Enabled settings, there is a new setting which is a dropdown to let me choose which of the pre-configured mqtt servers I want to use for the channel.

This should work for android and ios.

andrekir, 2024-03-14:
see #3296

djoris, 2024-04-16:
I can understand this request from a user perspective.
Another use case I imagine is having sensor data on a local MQTT with JSON for parsing in Home Assistant.

However, I think connecting the node to multiple MQTT brokers (servers) would be resource-intensive. Leading to time-outs or at least more power usage. 
The desired outcome could probably be best realised by using a private MQTT broker that bridges to a public MQTT broker. I know of two Dutch brokers like that, bridging certain topics to the public MQTT.

CamFlyerCH, 2024-04-17:
Regarding Home Assistant: I implemented a bridge the other way. I bridge msh/CH/2/json/# to my MQTT server from Home Assistant. Works fine to receive the telemetry from my routers.

closed 2024-04-28 by garthvh

andreasjohannesholy-bit, 2026-02-06:
As I am a stranger here in a foreign country, I want to use the Meshtastic server and also the one the locals here are using.

As a workaround I use two gateway nodes for different MQTT Servers, side by side, what also sometimes leads to problems, especially missing or loosing messages or delivers them double ...

As MQTT is only interesting when Internet (and electricity) is up, I also use gateway nodes powered by line. If power or internet (Routers) are down, I will not need the gateways. So the power is not really the problem.

Maybe there is a solution if there will be a gateway role for these cases.

By being managed in one node maybe also could manage the double delivery issue, when the same messages come over different servers.

Just an idea ...