meshtastic/firmware #3330: [Bug]: Flashing most recent firmware causes boot-loop Lilygo TTGO LoRa32 V2.1_1.6

CT5, 2024-03-03:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

firmware-tlora-v2-1-1_6-2.2.23.5672e68-update.bin

### Description

I'm new to LoRa, and am having some issues with my Lilygo TTGO LoRa32 V2.1_1.6. I'm able to firmware-tlora-v2-1-1.6-1.2.17.bin, but this is too old to communicate with Meshtastic Android app. Whenever I flash to any of the most recent alpha or stable releases, I get the output below log output. I've tried CLI, but am unable to get the device-update.bat command to work. Not sure what else to do to get this updated.

I also have a Heltec v3 that's working great, but for some reason, the TTGO is giving me a hard time.

### Relevant log output

```Shell
Using 'COM5' as serial port.
Writing at 0x001fc4f6... (100 %)Wrote 2021600 bytes (1215873 compressed) at 0x00010000 in 27.2 seconds (effective 594.5 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[16:34:06]ets Jun  8 2016 00:22:57
[16:34:06]
[16:34:06]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:06]configsip: 188777542, SPIWP:0xee
[16:34:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:06]mode:DIO, clock div:2
[16:34:06]load:0x3fff0018,len:4
[16:34:06]load:0x3fff001c,len:1044
[16:34:06]load:0x40078000,len:8896
[16:34:06]load:0x40080400,len:5828
[16:34:06]entry 0x400806ac
[16:34:06]ets Jun  8 2016 00:22:57
[16:34:06]
[16:34:06]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:06]configsip: 188777542, SPIWP:0xee
[16:34:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:06]mode:DIO, clock div:2
[16:34:06]load:0x3fff0018,len:4
[16:34:06]load:0x3fff001c,len:1044
[16:34:06]load:0x40078000,len:8896
[16:34:06]load:0x40080400,len:5828
[16:34:06]entry 0x400806ac
[16:34:07]ets Jun  8 2016 00:22:57
[16:34:07]
[16:34:07]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:07]configsip: 188777542, SPIWP:0xee
[16:34:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:07]mode:DIO, clock div:2
[16:34:07]load:0x3fff0018,len:4
[16:34:07]load:0x3fff001c,len:1044
[16:34:07]load:0x40078000,len:8896
[16:34:07]load:0x40080400,len:5828
[16:34:07]entry 0x400806ac
[16:34:08]ets Jun  8 2016 00:22:57
[16:34:08]
[16:34:08]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:08]configsip: 188777542, SPIWP:0xee
[16:34:08]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:08]mode:DIO, clock div:2
[16:34:08]load:0x3fff0018,len:4
[16:34:08]load:0x3fff001c,len:1044
[16:34:08]load:0x40078000,len:8896
[16:34:08]load:0x40080400,len:5828
[16:34:08]entry 0x400806ac
[16:34:09]ets Jun  8 2016 00:22:57
[16:34:09]
[16:34:09]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:09]configsip: 188777542, SPIWP:0xee
[16:34:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:09]mode:DIO, clock div:2
[16:34:09]load:0x3fff0018,len:4
[16:34:09]load:0x3fff001c,len:1044
[16:34:09]load:0x40078000,len:8896
[16:34:09]load:0x40080400,len:5828
[16:34:09]entry 0x400806ac
[16:34:09]ets Jun  8 2016 00:22:57
[16:34:09]
[16:34:09]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:09]configsip: 188777542, SPIWP:0xee
[16:34:09]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:09]mode:DIO, clock div:2
[16:34:09]load:0x3fff0018,len:4
[16:34:09]load:0x3fff001c,len:1044
[16:34:09]load:0x40078000,len:8896
[16:34:09]load:0x40080400,len:5828
[16:34:09]entry 0x400806ac
[16:34:10]ets Jun  8 2016 00:22:57
[16:34:10]
[16:34:10]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:10]configsip: 188777542, SPIWP:0xee
[16:34:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:10]mode:DIO, clock div:2
[16:34:10]load:0x3fff0018,len:4
[16:34:10]load:0x3fff001c,len:1044
[16:34:10]load:0x40078000,len:8896
[16:34:10]load:0x40080400,len:5828
[16:34:10]entry 0x400806ac
[16:34:11]ets Jun  8 2016 00:22:57
[16:34:11]
[16:34:11]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:34:11]configsip: 188777542, SPIWP:0xee
[16:34:11]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:34:11]mode:DIO, clock div:2
[16:34:11]load:0x3fff0018,len:4
[16:34:11]load:0x3fff001c,len:1044
[16:34:11]load:0x40078000,len:8896
[16:34:11]load:0x40080400,len:5828
[16:34:11]entry 0x400806ac
Serial port closed!
```

caveman99, 2024-03-04:
can you add a photo of the board? Also how are you trying to flash it?

thebentern, 2024-03-06:
> Firmware Version
> firmware-tlora-v2-1-1_6-2.2.23.5672e68-update.bin
> I'm able to firmware-tlora-v2-1-1.6-1.2.17.bin

You did an update instead of clean wipe and install. Updates don't work between major versions. You must do a full reinstallation.

closed 2024-03-06 by thebentern