meshtastic/firmware #3336: [Feature Request]: MQTT Location only mode

nukemrules, 2024-03-05:
### Platform

NRF52, ESP32

### Description

While the MQTT platform works great if you know what you are doing, lots of the new Meshtastic users are using the MQTT uplink only to get their location on one of the map sites. Unfortunately this resolves in many loops and multiple connections in a mesh to the official MQTT server. With the duty cycle of only 10% in Europe, this makes that the airtime is reached quite fast if you connect to the big topics on the official MQTT server too.

A way to move new users to a more secure way of publishing their or mesh locations is to have a separate topic on the MQTT server and a setting in the node to only publish its location. So its a one way publish. The node itself never reads any information from that topic. The Node map/list on the node should only be filled with nodes directly from the mesh. 
Device and/or Environment metrics could also be a big plus to publish in this or separate topic, but it should be optional. This way the node can be publicly monitored over the internet.

With this feature the mesh can do its thing without all of the extra traffic, but the rest of the world can see where the nodes/mesh networks are available.

GUVWAF, 2024-03-05:
Just to be sure: you know you can already only enable uplink (and not downlink) to avoid transmitting packets from MQTT over LoRa while still publishing your position on the map? 

I don't see why you need separate topics for position and telemetry to accomplish what you want.

nukemrules, 2024-03-05:
> Just to be sure: you know you can already only enable uplink (and not downlink) in the channel settings to avoid transmitting packets from MQTT over LoRa while still publishing your position on the map?
> 
> I don't see why you need separate topics for position and telemetry to accomplish what you want.

Sure I know, that is a full one way publish and means that everything from the mesh is published to the MQTT server too. Unfortunately when someone subscribes to that topic and uses up- and downlink, messages can't ever reach the nodes via the node that has only uplink enabled. With a location only mode, you protect the expectation of reaching nodes that can't be reached. And with a separate topic, you protect the mesh from flooding too.

GUVWAF, 2024-03-05:
> You can't hide messages in your public primary channel. Sure you can create a "private" channel, but you want a public one too for people from outside your mesh.

You can create a private primary channel to publish your position and not use it for anything else and have the "public" channel as secondary.
If you want to talk in the clear over LoRa on the primary channel but not publish it to MQTT, while still publishing your location, it sounds like a solution would be to allow periodic position publishing to MQTT on a different channel than the primary.

nukemrules, 2024-03-05:
> If you have a separate topic for position only, maps will not be able to show a short and long name, because they need the NodeInfo packets for that.

Sure, if that is needed too then NodeInfo should also be published in the same topic.

This could go hand in hand with issue [https://github.com/meshtastic/firmware/issues/3116]#3116(url), but needs an approach to make it easier for new users. So an out of the box approach or single click would help.

pdxlocations, 2024-03-05:
Chiming in only because I find this interesting.  I don't this feature as being practical, BUT I do see it as something lots of people have been asking for, and I'd prolly end up turning it on too.  It would be an easier way to let everyone know you're around, without sharing your entire mesh online.

Perhaps a solution would be a new module/portnum that sends out a combination of Long/Short name + Position information.  This portnum would NEVER TX over LoRa, only MQTT.

Danie10, 2024-03-06:
We are already using a country topic for MQTT for South Africa as msh/ZA and that is working well to filter more for local users, and still uploads our location to the map. So anyone can already choose their own topic.

GUVWAF, 2024-03-11:
This is implemented with #3365 and will be in the upcoming 2.3.0 release. Please ask the maintainer of the map you're using to incorporate the new `MapReport` packet on the `/2/map/` topic which will contain all kinds of information about the node (also e.g. firmware version and number of active local nodes).

closed 2024-03-11 by GUVWAF