meshtastic/firmware #3338: [Bug]:

davis-gotenna, 2024-03-06:
### Category

Hardware Compatibility

### Hardware

Rak11200

### Firmware Version

2.2.23.567e68, 2.2.16.1c6acfd

### Description

I'm experiencing a boot loop with my WisBlock setup.

Using RAK19007 base board with the RAK11200 ESP32 core and the RAK13300 LoRa module. After reset, the device seems to be stuck in a boot loop with meshtastic --info showing the reboot count constantly climbing. I tried firmware versions 2.2.16 and 2.2.23 and see the same thing with both.

The logs show the error right before reboot

### Relevant log output

```Shell
DEBUG | 13:29:57 33 [RadioIf] SX126x standby failed with error -2

assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:214 (err == RADIOLIB_ERR_NONE)

Backtrace: 0x40084469:0x3ffd93d0 0x40096dd1:0x3ffd93f0 0x4009d2a9:0x3ffd9410 0x400e217e:0x3ffd9540 0x400e1acc:0x3ffd9560 0x400e6495:0x3ffd9580 0x400e6550:0x3ffd95a0 0x402202de:0x3ffd95c0 0x400d5a3d:0x3ffd95e0 0x400d5b0a:0x3ffd9600 0x40104215:0x3ffd9620 0x400e00ea:0x3ffd9650 0x40137a4d:0x3ffd9670

ELF file SHA256: d45acb5d0d9deab8

E (36884) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (834) esp_core_dump_flash: No core dum�␆␄��ѥѥ���found!
E (834) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
��␀␂@␁INFO  | ??:??:?? 0
```

garthvh, 2024-03-10:
The RAK11200 only works with the 5005 base board

andyshinn, 2024-03-18:
I notice there is a `SX126X_ANT_SW`. In `variant.h` for this board i set `#define SX126X_ANT_SW WB_IO2` and it starts up the radio and it is found. Unfortunately, I still have other failures with this combo that I am trying to figure out.

erikdewildt, 2024-04-24:
@wal-stefan 

> I was guessing that the supply power for the RAK13300 on the IO-slot (3V3_S on pin 6) might not be enabled. Just my observation: connecting a pull-up resistor between RAK19007 pin header IO2 (which in controlling 3V3_S) and VDD seems to fix it.

I'm not that technical but I have bought a couple of these boards so I'm desperately looking for a fix.  I have made a small drawing of how I understand your "fix". Can you confirm that this is what you mean?

[image]

Thanks!

wal-stefan, 2024-04-24:
Hi @erikdewildt,

Yes, this was my fist attempt to handle my problem. And it seemed to work.

Currently, as being more a software guy than a hardware guy, I added two lines to source code of `main.cpp` to hopefully achieve the same. And it seems to work for a couple of weeks now.

> [!WARNING]
> But, of cause, when you add this, NEVER flash tis firmware to a hardware different to the one described!

...
```
    LOG_INFO("\n\n//\\ E S H T /\\ S T / C\n\n");

    initDeepSleep();

    // power on peripherals

    // RAK 19007 WisBlock Base: enable power on 3V3_S for SLOT A-D and IO-SLOT
    pinMode(WB_IO2, OUTPUT);
    digitalWrite(WB_IO2, 1);

#if defined(TTGO_T_ECHO) && defined(PIN_POWER_EN)
    pinMode(PIN_POWER_EN, OUTPUT);
    digitalWrite(PIN_POWER_EN, HIGH);
    // digitalWrite(PIN_POWER_EN1, INPUT);
#endif
```
...

Sorry, currently I do not have the time to figure out how this can be integrated into the Meshtasic firmware without breaking anything.

erikdewildt, 2024-04-24:
Thanks! 

I'll try the hardware fix for now.  

As the RAK5005-O is not available anymore and the RAK11200 is the only RAK option which includes WiFi I assume more people would like to use the RAK11200 + RAK19007 combination.

Would it be an option to have a dedicated RAK11200 + RAK19007 and a RAK11200 + RAK5005-O firmware? This way both combinations could be supported if I understand correctly.

But I'm not into all details, just suggesting a possible option.

Anyway thanks again for your work on this project!

Regards,
Erik

andyshinn, 2024-04-24:
> ```
>     LOG_INFO("\n\n//\\ E S H T /\\ S T / C\n\n");
>
>     initDeepSleep();
>
>     // power on peripherals
>
>     // RAK 19007 WisBlock Base: enable power on 3V3_S for SLOT A-D and IO-SLOT [...]

This is what my comment was referring to. That code is pretty much already there under a condition for if `SX126X_ANT_SW` is set. If you add `#define SX126X_ANT_SW WB_IO2` to the `variant.h` file then it will also do the same thing.

williamtrelawny, 2024-06-11:
So do I understand correctly that the RAK11200 core does work with RAK19007 but only with the 2 addtl lines above?

wal-stefan, 2024-06-15:
I observed the problem, and the solution may only be valid, on the combination of these three components:
1. RAK19007 WisBlock Base Board 2nd Gen
2. RAK11200-C WiFi and BLE Espressif ESP32-WROVER
3. RAK13300 LoRaWAN SX1262
I also observed, RAK is changing significant things on their boards from revision to revision, without any notice.
So, problem may exist, and solution may be valid only on certain revisions of these boards.

The reason seems to be, that the RAK13300 LoRaWAN SX1262 needs power on the slot's 3V3_S pin to work.

The RAK19007 WisBlock Base Board can enable or disable this power (probably to save power). The Base Board's 3V3_S-power-enable signal is connected to the pin-header AND the CPU-slot (So better, do not use both to enable. The high ohm resistor should be on the safe side).
When a RAK11200-C (running Meshtastic) is plugged in the CPU-slot it seems not to enable the 3V3_S power. And the Base Board does not neither. As the enable signal may even be floating (undefined) it may even work sometimes, this way.
Better intentionally enable 3V3_S, either from the Base Board or the CPU.

bls8486, 2024-07-09:
Would you be able to share your modified compiled code?

sgoudelis, 2024-07-09:
I am also interested in this combination, is there a chance that it will be included in the main trunk ?

bls8486, 2024-07-09:
> Would you be able to share your modified compiled code?

Also, how did you flash it?  It has disappeared in the web flasher and I'm having issues using other esp32 flashers.

bls8486, 2024-07-11:
> This is what my comment was referring to. That code is pretty much already there under a condition for if `SX126X_ANT_SW` is set. If you add `#define SX126X_ANT_SW WB_IO2` to the `variant.h` file [...]

Made these changes and compiled with 2.3.16.  Made just these changes.  Loaded to a Rak 11200, 19007, and 13300.  Seems to be functioning.  I think we should ask for this variant to be added.  Have any of you had any issues?

PR #4690 (2024-09-12): Define SX126X_ANT_SW for the RAK11200 on the RAK19007 base
Fixes #3338, allows the RAK13300 and GNSS module to function correctly when used on the RAK19007 base with the RAK11200 core module by activating the 3.3V power rail correctly.