meshtastic/firmware #3345: [Bug]: Improved Display Compatibility for Heltec Wireless Stick V3 (64x32) - Bluetooth Code Display

JanisPlayer, 2024-03-07:
### Category

BLE

### Hardware

Heltec V3

### Firmware Version

2.2.25.9d37a8d1

### Description

I noticed that the Heltec wireless-stick-v3 has a display error. This seems to be because the information on the manufacturer's website is incorrect. The page: https://heltec.org/project/wireless-stick-v3/ states that it has a 128x64 pixel display, but it is actually a 64x32 pixel display.

I made some changes to https://github.com/meshtastic/firmware/blob/master/src/graphics/Screen.cpp, and at least I was able to display the Bluetooth code. Are there any plans for adapting to such small screens on your end? I am aware that this would involve displaying texts on small screens in a special way, such as moving texts, and programming the recognition of these displays. I have adapted the code on my end so that, for now, only the Bluetooth code is displayed correctly. Such an adjustment is possible with the current code without as many changes as I made.

My test changes:

```cpp
static void drawFrameBluetooth(OLEDDisplay *display, OLEDDisplayUiState *state, int16_t x, int16_t y)
{
    int x_offset = display->width() / 2;
    int y_offset = display->height() <= 32 ? 0 : 16;
    display->setTextAlignment(TEXT_ALIGN_CENTER);
    display->setFont(FONT_SMALL); 

    display->drawString(x_offset + x, y_offset + y, "Bluetooth");

    y_offset += FONT_HEIGHT_SMALL;

    display->drawString(x_offset + x, y_offset + y, "Enter this code");

    String displayPin(btPIN);
    String pin = displayPin.substring(0, 3) + " " + displayPin.substring(3, 6);
    y_offset += FONT_HEIGHT_SMALL - 5;

    display->drawString(x_offset + x, y_offset + y, pin);

    String deviceName = "ID: ";
    deviceName.concat(getDeviceName());
    y_offset += FONT_HEIGHT_SMALL - 5;

    display->drawString(x_offset + x, y_offset + y, deviceName);
}
```

Issue: https://meshtastic.discourse.group/t/heltec-wireless-stick-v3-with-0-49-oled/9158/7

caveman99, 2024-03-08:
I have one of these incoming and will take a stab at it. Thanks for the initial work. No promises though :-)

JanisPlayer, 2024-03-09:
@caveman99  Thank you for the information; now I know that it's probably normal with the resolution. It seems like the product description and documentation might be incorrect.

That's cool! If it works out and they manage to do it, I'll be excited. I've found alternative fonts that can further reduce the text size. With such a small display resolution, displaying only the essentials is necessary, and dealing with long texts requires shortening or making the text move. It's indeed a complex task; I wish you much success. But I believe as long as the Bluetooth code is displayed, it's already sufficient. :)

DrBlackross, 2024-05-30:
putting battery life, node count and or signal strength on other screen that are toggle-able with the usr button would be cool

(trying to find screen settings in the rest of the code is a bit tricky so far)