meshtastic/firmware #3376: [Bug]: T-Beam 1.1 just turns off after first config

DooMMasteR, 2024-03-11:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.2.24.e6a2c06

### Description

I can flash the firmware just fine, but once I configure it to just ROUTER + EU_868 and save, it reboots and then "turns off" with the following messages on serial:

and after this I cannot get it to work (normally).

The issue is, that the devices just goes to sleep and then cannot be used via USB/Serial/Bluetooth anymore it works fine after waking it with the user key and quickly disabling the power saving, by "enabling" it since the defaults won't let it sleep fast enough.

I think this is a "bad" default.

### Relevant log output

```Shell
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=1, isCharging=1, batMv=4207, batPct=100
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
INFO  | ??:??:?? 4 [GPS] GNSS module configuration saved!
INFO  | ??:??:?? 4 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [PowerFSM] Enter state: ON
INFO  | ??:??:?? 4 [PowerFSM] Initialise the NimBLE bluetooth module
DEBUG | ??:??:?? 5 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | ??:??:?? 5 [GPS] No GPS lock
DEBUG | ??:??:?? 5 [GPS] onGPSChanged() pos@0, time=0, lat=0, lon=0, alt=0
INFO  | ??:??:?? 5 [GPS] updatePosition LOCAL pos@0, time=0, latI=0, lonI=0, alt=0
DEBUG | ??:??:?? 5 [GPS] Setting local position: latitude=0, longitude=0, time=0
DEBUG | ??:??:?? 5 [GPS] Node status update: 1 online, 1 total
DEBUG | ??:??:?? 8 [PowerFSM] Enter state: SDS
INFO  | ??:??:?? 8 [PowerFSM] Entering deep sleep for 86400 seconds
INFO  | ??:??:?? 8 [PowerFSM] Disable bluetooth
INFO  | ??:??:?? 8 [PowerFSM] Saving /prefs/db.proto
INFO  | ??:??:?? 8 [PowerFSM] Saving /prefs/config.proto
INFO  | ??:??:?? 9 [PowerFSM] Saving /prefs/module.proto
INFO  | ??:??:?? 9 [PowerFSM] Saving /prefs/channels.proto
INFO  | ??:??:?? 10 [PowerFSM] Setting GPS power=0
E (10781) RTCIO: rtc_gpio_pullup_en(109): RTCIO number error
E (10781) RTCIO: rtc_gpio_pullup_en(109): RTCIO number error
```
```

caveman99, 2024-03-11:
Please refer to https://meshtastic.org/docs/configuration/radio/device/#role-comparison

closed 2024-03-11 by caveman99

DooMMasteR, 2024-03-11:
Well, that's still an unusable deadlock, that comes very unexpected when following the config guide
a serial hold-off of sleep-lock would be the minimum I guess...  or at least a clear warning that setting "ROUTER" should basically be the last setting ever set.