meshtastic/firmware #3384: 🚀 [Feature Request]: protecting admin channel

AssetBurned, 2024-03-12:
### OS

iOS, iPadOS, macOS

### Description

Given that the “admin” channel is such a critical thing, there should be some clearly visible warning if the channel is setup without a key.

I don’t think that completely blocking it would be good, as it could be useful to demonstrate the functionality.

### Participation

- [ ] I am willing to submit a pull request for this issue.

thebentern, 2024-03-12:
Can't be enforced on the firmware without blocking the request altogether

AssetBurned, 2024-03-14:
No idea why this is moved over from the application into the firmware. The Apple App already has other warnings and even blocks things. For example the web client lets me set a key for a channel consisting of 16 letters & digits for example "DaChanne7K3yH3re", while the AppleApp doesn't accept this. So the suggestion here was to provide the user of the App with an popup, overlay or just an text under the password field along the line of "Hey you did not setup a key for this channel, so everyone can change settings. Is that really what you want?"
This is a functionality that should be easily be implementable within the app before the save will send the change to the actual device running the Meshtastic firmware.