meshtastic/firmware #3389: [Bug]: Heltec V3.1 says battery powered when connected to USB-C power

Calimerorulez, 2024-03-12:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.0

### Description

Thank you for your continuing work in developing Meshtastic.

When having USB-C power connected to my Heltec V3.1, it shows the battery logo on screen, and the serial console debug shows:

`DEBUG | ??:??:?? 61 [Power] Battery: usbPower=0, isCharging=0, batMv=3886, batPct=69`

### Relevant log output

```Shell
DEBUG | ??:??:?? 61 [Power] Battery: usbPower=0, isCharging=0, batMv=3886, batPct=69
```

caveman99, 2024-03-12:
you need to calibrate your ADC. Not a bug. See https://meshtastic.org/docs/configuration/radio/power/#calibration-process-attribution

closed 2024-03-12 by caveman99

Calimerorulez, 2024-03-12:
Thanks @caveman99

caveman99, 2024-03-12:
Yes because the end voltage of the battery also determines the usb display