meshtastic/firmware #3390: [Bug]: T-Echo wo't boot with 2.3.0

pista01, 2024-03-12:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.2.24.e6a2c06 Beta

### Description

I have 2 T-Echo devices (v1 boards) that refuse to boot after installing 2.3.0.  Multiple boot attempts all freeze at the same place.

### Relevant log output

```Shell
//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  module.proto (98 Bytes)
DEBUG | ??:??:?? 0  channels.proto (140 Bytes)
DEBUG | ??:??:?? 0  config.proto (89 Bytes)
DEBUG | ??:??:?? 0  db.proto (979 Bytes)
DEBUG | ??:??:?? 0 Using analog input 4 for battery level
INFO  | ??:??:?? 0 Scanning for i2c devices...
DEBUG | ??:??:?? 0 Scanning fo
```

Mictronics, 2024-03-12:
Can't confirm. Tried 2.3.0.5f47ca1 Alpha and also the latest master ee685b4. Both work on one of my T-Echos with board marked v1.0 2021-06-28. The board is equipped with BME280.

pista01, 2024-03-12:
Maybe it's older than that.  These are one of the first ones they shipped.  I don't recall that there was an option for a BME280 back then.

pista01, 2024-03-12:
I checked that already when I opened issue #3249.  The test build of 2.2.25 from that issue worked.  Nothing since.

thebentern, 2024-03-14:
@pista01 can you try this one?
https://github.com/meshtastic/firmware/pull/3406

andrekir, 2024-03-14:
PR#3406 .uf2 for the T-echo here: https://github.com/meshtastic/firmware/actions/runs/8274137437/artifacts/1324540663

but please try this one first: https://github.com/meshtastic/firmware/actions/runs/8279238369/artifacts/1325679915

pista01, 2024-03-14:
I tried them both.  Both are working.

jeremybox, 2024-03-14:
Both versions linked above still seem to freeze my device.

`@INFO  | ??:??:?? 1

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1 Using analog input 4 for battery level
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1`

andrekir, 2024-03-15:
@jeremybox did you remove the file names under your first line? it should look like:

```
DEBUG | ??:??:?? 2 Filesystem files:
DEBUG | ??:??:?? 2  db.proto (331 Bytes)
DEBUG | ??:??:?? 2  config.proto (89 Bytes)
DEBUG | ??:??:?? 2  module.proto (113 Bytes)
DEBUG | ??:??:?? 2  channels.proto (97 Bytes)
(...)
```

could be a different issue. do you have any known working firmware version? or try flashing [SoftRF](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/binaries/nRF52840/SoftRF/MassStorage/SoftRF-firmware-v1.4-nRF52.uf2), it runs a POST check on startup.

jeremybox, 2024-03-15:
I did not edit the output.  I just tested and had no output at all under the SoftRF firmware.  I did the nrf erase and reflashed the meshtastic firmware and continue to have the blank filesystem files ouput.  If you think I"m experiencing a different issue, I can file a separate ticket.

closed 2024-03-15 by thebentern