meshtastic/firmware #3391: [Bug]: In the ROUTER role, the device reboots instead of waking up the CPU

Holyszewski, 2024-03-12:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.0.5f47ca1

### Description

_Hi, this is my first bug report, so please be forgiving of any newbie imperfections._

When I set my Heltec v3 to the ROUTER role, it stops working correctly. The RF module should wake up the sleeping CPU when a message arrives, but instead of waking up, the CPU always reboots.
I attached the log, note especially the lines: 
`INFO  | ??:??:?? 0 Booted, wake cause 3 (boot count 4), reset_reason=ext1 RTC_CNTL`
`INFO | ??:??:?? 0 Booted, wake cause 2 (boot count 5), reset_reason=ext0 RTC_IO`
`E (9200) RTCIO: rtc_gpio_isolate(171): RTCIO number error`
The problem is manifested in firmware versions 2.2.24 and 2.3.0. In previous versions 2.2.23 and 2.2.22 it works as it should.

### Relevant log output

```Shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (44) esp_core_dump_flash: No core dump partition found!
E (44) esp_core_dump_flash: No core dump partition found!
”Ă @INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 3 (boot count 4), reset_reason=ext1 RTC_CNTL
DEBUG | ??:??:?? 0 Filesystem files (524288/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (95 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (204 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (104 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-CcvK1znC.css.gz (16231 Bytes)
DEBUG | ??:??:?? 0  /static/index-siIur9KN.js.gz (234592 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BS7WETSR.js.gz (209824 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   436][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000
[   438][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
DEBUG | ??:??:?? 0 Using analog input 1 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   470][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 2
[   499][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
[... 359 lines cut ...]
INFO  | ??:??:?? 1 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 1 SX1262 Radio init succeeded, using SX1262 radio
INFO  | ??:??:?? 1 Not using WIFI
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms
DEBUG | ??:??:?? 1 LoRA bitrate = 118.500000 bytes / sec
INFO  | ??:??:?? 1 PowerFSM init, USB power=0
DEBUG | ??:??:?? 1 Enter state: BOOT
[  1217][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz
DEBUG | ??:??:?? 1 [Power] Battery: usbPower=0, isCharging=0, batMv=3100, batPct=0
DEBUG | ??:??:?? 1 [Screen] Screen: Started...
INFO  | ??:??:?? 1 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [PowerFSM] Enter state: ON
INFO  | ??:??:?? 4 [PowerFSM] Initialise the NimBLE bluetooth module
INFO  | ??:??:?? 5 [PowerFSM] Turning off screen
DEBUG | ??:??:?? 7 [PowerFSM] Enter state: SDS
INFO  | ??:??:?? 7 [PowerFSM] Entering deep sleep for 86400 seconds
INFO  | ??:??:?? 7 [PowerFSM] Disable bluetooth
INFO  | ??:??:?? 7 [PowerFSM] Saving /prefs/db.proto
INFO  | ??:??:?? 8 [PowerFSM] Saving /prefs/config.proto
INFO  | ??:??:?? 8 [PowerFSM] Saving /prefs/module.proto
INFO  | ??:??:?? 8 [PowerFSM] Saving /prefs/channels.proto
E (9200) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (9201) RTCIO: rtc_gpio_isolate(171): RTCIO number error
E (9201) RTCIO: rtc_gpio_isolate(171): RTCIO number error
```

thebentern, 2024-03-14:
@Holyszewski can you try the build assets for this firmware change?
https://github.com/meshtastic/firmware/pull/3405

closed 2024-03-14 by thebentern

Holyszewski, 2024-03-19:
> @Holyszewski can you try the build assets for this firmware change? #3405
> 
> Here's the Heltec V3 zip: https://github.com/meshtastic/firmware/actions/runs/8273520266/artifacts/1324406388

I tried the build you provided. It solved the original problem, but a new one appeared instead: the dormant router doesn't respond to button presses.

PR #3405 (2024-03-14): Try-fix router missed messages
Closes #3391