meshtastic/firmware #3396: [Bug]: multiple unrelated nodes reporting the same location

Bamorph, 2024-03-13:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6, Heltec V3, Other

### Firmware Version

2.2.23

### Description

On the ANZ mqtt server, Multiple nodes 100s of KMs apart with fixed locations are all reporting the same incorrect location.

Has happened a few times in the last week. In different, locations. But seems to be attracted to an original node. As an example, the other day a few nodes popped up next to my house, from a different state.

This is an example of some nodes that have had the issue: it is an assortment of hardware and firmware.

```
!da5439c8 | ROUTER_CLIENT | HELTEC_V3
!fa672e6c | ROUTER_CLIENT | HELTEC_WSL_V3
!da56bcf0 | ROUTER_CLIENT | HELTEC_V3
!95a587f8 | CLIENT | TLORA_V2_1_1P6
!75f17fd0 | ROUTER_CLIENT | TLORA_T3_S3
!da56a78c | ROUTER_CLIENT | HELTEC_V3
!fa672e6c | ROUTER_CLIENT | HELTEC_WSL_V3
!da541158 | ROUTER_CLIENT | HELTEC_V3
!7a6c6e04 | ROUTER_CLIENT | TBEAM
```

[image]

GUVWAF, 2024-03-13:
Do you know if those nodes have JSON enabled and might be running firmware <2.2.20? There was a bug where the "sendposition" JSON message caused *all* nodes receiving this to send out this position packet, but this was fixed by #3141.

thebentern, 2024-03-13:
Are we sure this isn't an issue with the Android client?

GUVWAF, 2024-03-14:
> They do not have Json enabled. And are running firmware 2.2.23

That seems to rule out my suspicion. Would you be able to check with the [CLI ](https://meshtastic.org/docs/software/python/cli/installation/) command `meshtastic --nodes` if you see the same there?

closed 2024-03-19 by thebentern