meshtastic/firmware #3400: Better error handling for radiolib errors

jp-bennett, 2024-03-13:
### Platform

ESP32

### Description

Currently RadioLib errors are just an INFO message. We should check for errors, and make this an ERROR message.

fifieldt, 2024-11-16:
In the code, you'll often see checks for RADIOLIB_ERR_NONE. To solve this issue, read up on the RadioLib error codes: https://jgromes.github.io/RadioLib/group__status__codes.html

and then make logic that handles errors appropriately for the situation.

You can see an example in src/mesh/SX128xInterface.cpp for RADIOLIB_ERR_INVALID_FREQUENCY

rostekus, 2025-02-20:
I want to tackle this issue but first i want to clarify what should be done.

1. Addiotional error handling for RadioLib

Hypothetical example:

```cpp
// src/mesh/SX126xInterface.cpp:147  
       
    int err = lora.readRegister(SX126X_REG_CRC_POLYNOMIAL_LSB, &crcLSB, 1);
    if(err != RADIOLIB_ERR_NONE)
        RECORD_CRITICALERROR(CriticalErrorCode_SX1262Failure);
}
```
We want to have:

```cpp
    int err = lora.readRegister(SX126X_REG_CRC_POLYNOMIAL_LSB, &crcLSB, 1);
    if(err != NOT_CRITICAL_RADIOLIB_ERR)
        LOG_WARN("Warning"); //Just an example
    else if (err != RADIOLIB_ERR_NONE)
        RECORD_CRITICALERROR(CriticalErrorCode_SX1262Failure);
}
```

If it is possible please assign me to this issue

fifieldt, 2025-02-20:
Thanks @rostekus !

So,  right now we make calls to radiolib and we check for err != RADIOLIB_ERR_NONE which is great - it means you have no errors.. However, the major issue is we don't check for the error cases, and so the only feedback we can give to the user is "there were not no errors" :) 

I think an example of where we're doing something other than the 'basic' handling at the moment is src/mesh/SX128xInterface.cpp line 72 . In the below code, we try to start a radio, print the result. If the result is that we chose a bad frequency that the radio doesn't support, then we tell the user and actually try to recover. If everything is fine, we continue with our work.

```
int res = lora.begin(getFreq(), bw, sf, cr, syncWord, power, preambleLength);
LOG_INFO("SX128x init result %d", res);

    if ((config.lora.region != meshtastic_Config_LoRaConfig_RegionCode_LORA_24) && (res == RADIOLIB_ERR_INVALID_FREQUENCY)) {
        LOG_WARN("Radio only supports 2.4GHz LoRa. Adjusting Region and rebooting");
        config.lora.region = meshtastic_Config_LoRaConfig_RegionCode_LORA_24;
        nodeDB->saveToDisk(SEGMENT_CONFIG);

...

    if (res == RADIOLIB_ERR_NONE)
        res = lora.setCRC(2);

    if (res == RADIOLIB_ERR_NONE)
        startReceive(); // start receiving

    return res == RADIOLIB_ERR_NONE;
```

The logic to try to recover is not necessary to fix this bug. However, checking the return from lora.begin against the [possible](https://jgromes.github.io/RadioLib/group__status__codes.html) radiolib errors and giving an appropriate error message would be amazing.

An example of our typical approach we can probably improve is found in the same file at around line 150. Here we try to set the power for a radio, and if anything at all goes wrong we fail and give the user a generic error.  However, radiolib actually has RADIOLIB_ERR_INVALID_OUTPUT_POWER ... which we could use to provide more information for the user.

```
err = lora.setOutputPower(power);
    if (err != RADIOLIB_ERR_NONE)
        LOG_ERROR("SX128X setOutputPower %s%d", radioLibErr, err);
    assert(err == RADIOLIB_ERR_NONE);
```

So, your example is basically correct - only "err != RADIOLIB_ERR_NONE" would be a success case, not a failure case .. but I think that's just a typo. :) Search the code for anywhere we do "err != RADIOLIB_ERR_NONE" and you'll probably find something that can be improved :)

closed 2025-05-26 by garthvh