meshtastic/firmware #3408: TextMessageModule auto response after message parsing

Kralg, 2024-03-14:
Hello, 
I would like to make a custom parser in _ProcessMessage TextMessageModule::handleReceived(const meshtastic_MeshPacket &mp)_
Can somebody give me some thoughts is it possible after parsing to send back a message to the sender of the message from this method?

garthvh, 2024-03-14:
https://github.com/meshtastic/chirpy

Kralg, 2024-03-14:
Thanks @garthvh Is it possible the link does not work? 

I also looked at the detectionSensorModule. I am thinking maybe to take _void DetectionSensorModule::sendCurrentStateMessage()_ and reuse it in the _TextMessageModule_. At least one additional thing that I see differs is that this Detection class inherits from _private concurrency::OSThread_ can you please write is this related to using send functions?

closed 2024-03-14 by Kralg