meshtastic/firmware #3429: [Bug]:

eldoctorcool, 2024-03-17:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.3.0

### Description

Traceroutes don't register on the Android app. They work with fw 2.2.24. but not with 2.3.0

eldoctorcool, 2024-03-17:
Android app is 2.2.24.

eldoctorcool, 2024-03-17:
Well, I've managed to get a Traceroute to pop up after trying 18 times... this never happened before.

eldoctorcool, 2024-03-17:
I am able to traceroute from a 2.3.0 node to any of my 2.2.24 fw nodes just fine, but if I try anything that has fw 2.3.0 from a 2.3.0 node it simply won't show a traceroute.

eldoctorcool, 2024-03-17:
Repro steps: 
-Need 3 nodes and Android version 2.2.24.
-Install 2.3.0 on two nodes.
-Install 2.2.24 on one node.
-From the 2.3.0 node send a Traceroute to the 2.2.24. You'll see the Traceroute register.
-From the 2.3.0 node send a Traceroute to the 2.3.0 node, Traceroutes won't register at all on the app.

GUVWAF, 2024-03-18:
Maybe this is related to https://github.com/meshtastic/Meshtastic-Android/issues/916, but I am unable to reproduce. Traceroutes work for me with a node using firmware 2.2.24 and another 2.3.0, and also between two 2.3.0 nodes.

eldoctorcool, 2024-03-18:
The interesting detail I noticed is that the node that was pinged goes to the top of the node list, like when packet comes from the mesh... but it doesn't acknowledge it as a Traceroute on the app... so, maybe this is an app issue? Packet garbled could be? Maybe we could test the hop count not changed as suggested on the other bug?

GiuStel, 2024-03-20:
Here with android app and fw on 2.3.x , doensn't matter the patch version, traceroutes always show up, but always show 1 hop only: the target node.

GUVWAF, 2024-03-21:
Let's continue this in #3452.

closed 2024-03-21 by GUVWAF