meshtastic/firmware #3430: [Bug]: LilyGo T3-S3 needs manual reset to enter bootloader mode

jurriaan, 2024-03-17:
### Category

Hardware Compatibility, Serial

### Hardware

Other

### Firmware Version

v2.3.0.5f47ca1

### Description

Related to https://github.com/meshtastic/firmware/issues/2308#issuecomment-1445136045

When updating the firmware through the `device-update.sh` script included in the firmware zip file I always need to manually enter boot mode on my LilyGo T3-S3.

I investigated why this was happening, and found out that this is due to the usage of TinyUSB for the serial port.

In order to reset the arduino to bootloader mode when using TinyUSB you first have to do a 1200 baud reset sequence as implemented using Python here:

https://gist.github.com/jurriaan/fec7d176cd4b27c15a534ac7014d337e#file-reset_serial-py-L1-L39

Running this script before `device-update.sh` allows me to flash the firmware without pressing buttons on the T3-S3.

While this works and something similar can probably be implemented in the device update script, a better way might be using the Serial-JTAG interface for serial communication instead of TinyUSB. See https://github.com/espressif/arduino-esp32/issues/8237#issuecomment-1563497918

prokrypt, 2024-03-26:
A simple `stty -F $DEVICE 1200` in the script may suffice, though we should probably require the user to specify the port/device...

garethhcoleman, 2024-04-21:
I've poked around with the two relevent switches ARDUINO_USB_MODE and ARDUINO_USB_CDC_ON_BOOT.

Many boards use the combination ARDUINO_USB_CDC_ON_BOOT=1 and ARDUINO_USB_MODE=0 which enables TinyUSB stack. I believe this is mainly because then the VID and PID of the device can be presented to the host, so you see something like  this:
[206677.891167] usb 2-1.2: New USB device found, idVendor=16d0, idProduct=1178, bcdDevice= 1.00
[206677.891202] usb 2-1.2: Product: unPhone 9

instead of this:
[206723.646047] usb 2-1.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[206723.646080] usb 2-1.2: Product: USB JTAG/serial debug unit

So I would advocate adding the invocation to the script and letting boards use tinyUSB if they wish - various boards use various combinations at present and changing these options underneath them would require a lot of testing to make sure things didn't break....

oseiler2, 2024-04-24:
A similar change has recently been done for the T-Beam S3 (#3597, #3652)
This should do: [tlora-t3s3-v1.patch](https://github.com/meshtastic/firmware/files/15092076/tlora-t3s3-v1.patch)

fifieldt, 2024-09-16:
The patches referenced in the previous comment didn't work out and had to be reverted.

The web flasher now has a button for 1200 baud reset.

I guess the fix here is to take code from the python script from the original report and add it to device-update.sh

fifieldt, 2024-09-16:
@jurriaan would you propose adding reset_serial.py to the bin directory in the firmware?

closed 2024-10-07 by thebentern