meshtastic/firmware #3431: [Feature Request]: "Who can see me" broadcast?

Slyke, 2024-03-17:
### Platform

ESP32

### Description

Was thinking about this for finding the optimal position to place a new node. It's basically a feature that makes your node send out a broadcast, asking "Who can see me?" And everyone who can, replies with their ID, how well they received the signal, and their location (if available), plus any other relevant info. This should be optionally (dis/en)abled by repliers, and there should be configurable rate limiting from both ends.

This broadcast should not be forwarded or rebroadcasted. Other nodes should only reply if they are of the role:
* Client
* Client Mute
* Router
* Router_Client

garthvh, 2024-03-17:
This already happens

closed 2024-03-17 by thebentern

Slyke, 2024-03-17:
@garthvh How to make it broadcast?

Slyke, 2024-03-18:
@rcarteraz is clearing the NodeDB the best way to produce a new broadcast to test a new location?

garthvh, 2024-03-18:
No need to do anything, it happens automatically

Slyke, 2024-03-18:
@garthvh this is specifically for determining the best place to place a Meshtastic Node. How would Meshtastic know that I've moved the node and want to now see who it can see? I don't understand how it would automatically know this, especially if it has already learned some nodes from previous attempts and may no longer be able to see them at a new location.

Slyke, 2024-03-18:
Awesome, thanks for the info!