meshtastic/firmware #3433: [Board]: New T-beam Supreme unable to operate on battery mode

Gitwhisky, 2024-03-18:
### SOC

ESP32

### Lora IC

ESP32-S3 LoRa Development Board with SX1262 868/915MHz GPS U-blox L76K

### Product Link

https://www.lilygo.cc/products/t-beamsupreme-m?_pos=5&_sid=546a32f19&_ss=r

### Description

Hey guys, I purchased 4 units of the above mentioned and separately purchased the recommended battery (Samsung25R 3.7V Rechargeable 18650 Battery 25R Flat Top, Real 2500mAh) and tested the batteries with a volt meter which showed good standing. Yet when I connect the batteries to the T-beam, a red light behind the screen appears showing power connection, but will not operate the T-beam at all on battery mode, will not turn on. I was able to download the software and set them up all up via USB charging, but the second I disconnect the USB, it will no power on and run on battery alone, on all 4 units.   Please advise.  Brian

battery link:https://www.amazon.com/dp/B09CQFCKLZ?psc=1&ref=ppx_yo2ov_dt_b_product_details

[image]

thebentern, 2024-03-18:
Please add more context to this issue.

Gitwhisky, 2024-03-18:
I tried to delete post, I was able to fix the problem. Since I couldn't
delete, I cleared all fields. I appreciate you looking into it.

closed 2024-03-19 by thebentern