meshtastic/firmware #3436: [Bug]: Crashing when activating PWM buzzer on Heltec V3 board

sdhuk, 2024-03-18:
### Category

Hardware Compatibility

### Hardware

Heltec V3

### Firmware Version

2.3.0.5f47ca1

### Description

Hello,

I've tried to add a buzzer to GPIO 7 on a Heltec v3. FW 2.3.0.  Seems to crash occasionally when triggered. Can't get the ringtone to work either. Setup via the cli:

```
meshtastic --set external_notification.enabled true --set external_notification.alert_bell_buzzer true --set external_notification.alert_message_buzzer true --set external_notification.output_buzzer true --set external_notification.use_pwm true --set external_notification.output_buzzer 7 --set external_notification.use_pwm true --set-ringtone 'd=4,o=6,b=100:32c,32p,32c7,32p,32a5,32p,32a,32p,32a#5,32p,32a#,2p,32c,32p,32c7,32p,32a5,32p,32a,32p,32a#5,32p,32a#,2p,32f5,32p,32f,32p,32d5,32p,32d,32p,32d#5,32p,32d#,2p,32f5,32p,32f,32p,32d5,32p,32d,32p,32d#5,32p,32d#'
```

When I receive a message, the buzzer activates, but the node will often crash with the following error in the log:

```
ERROR | ??:??:?? 28 [Router] Alloc an err=0,to=0x8ad1840,idFrom=0x1278cb1f,id=0x184cac83
```

Thanks,

Stuart

### Relevant log output

```Shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (349) esp_core_dump_flash: No core dump partition found!
E (349) esp_core_dump_flash: No core dump partition fo�� @INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (524288/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (146 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (88 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (416 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (123 Bytes)
DEBUG | ??:??:?? 0  /prefs/ringtone.proto (221 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-CcvK1znC.css.gz (16231 Bytes)
DEBUG | ??:??:?? 0  /static/index-siIur9KN.js.gz (234592 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BS7WETSR.js.gz (209824 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   702][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000
[... 374 lines cut ...]
DEBUG | ??:??:?? 46 encoding toPhone packet to phone variant=2, 53 bytes
INFO  | ??:??:?? 46 From Radio onread
DEBUG | ??:??:?? 46 [RadioIf] Completed sending (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=4 Ch=0x8 encrypted hopStart=4 priority=1)
DEBUG | ??:??:?? 51 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=44, time 559 ms
DEBUG | ??:??:?? 51 [RadioIf] Lora RX (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.75 rxRSSI=-55)
DEBUG | ??:??:?? 51 [RadioIf] AirTime - Packet received : 559ms
DEBUG | ??:??:?? 51 [Router] Rx someone rebroadcasting for us (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.75 rxRSSI=-55)
DEBUG | ??:??:?? 51 [Router] didn't find pending packet
DEBUG | ??:??:?? 51 [Router] Found existing packet record for fr=0xda5c88d8,to=0xffffffff,id=0x184cac85
DEBUG | ??:??:?? 51 [Router] Resending implicit ack for a repeated floodmsg
DEBUG | ??:??:?? 51 [Router] enqueuing for send (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6.75 rxRSSI=-55 hopStart=2)
DEBUG | ??:??:?? 51 [Router] txGood=3,rxGood=4,rxBad=0
DEBUG | ??:??:?? 51 [Router] rx_snr found. hop_limit:2 rx_snr:6.750000
DEBUG | ??:??:?? 51 [Router] rx_snr found in packet. Setting tx delay:2898
DEBUG | ??:??:?? 51 [Router] Found existing packet record for fr=0xda5c88d8,to=0xffffffff,id=0x184cac85
DEBUG | ??:??:?? 51 [Router] Add packet record (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.75 rxRSSI=-55)
DEBUG | ??:??:?? 51 [Router] Ignoring incoming msg, because we've already seen it (id=0x184cac85 fr=0xd8 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=6.75 rxRSSI=-55)
DEBUG | ??:??:?? 51 [Router] cancelSending id=0x184cac85, removed=1
DEBUG | ??:??:?? 51 [Router] Incoming message was filtered 0xda5c88d8
INFO  | ??:??:?? 61 [PositionModule] Sending pos@0:6 to mesh (wantReplies=1)
DEBUG | ??:??:?? 61 [PositionModule] Skipping location send because precision is set to 0!
DEBUG | ??:??:?? 61 [PositionModule] allocReply returned a nullptr
DEBUG | ??:??:?? 61 [Power] Battery: usbPower=0, isCharging=0, batMv=3957, batPct=76
DEBUG | ??:??:?? 81 [Power] Battery: usbPower=0, isCharging=0, batMv=4019, batPct=84
```

garthvh, 2024-03-18:
Does that pin support PWM?

sdhuk, 2024-03-19:
I think so. My understanding is that any of the GPIO pins on the ESP32-S3 can be used as PWM pins, with the exception of the input only pins. Happy to be corrected on that if I’m wrong! Thanks.

todd-herbert, 2024-03-23:
> INFO  | ??:??:?? 1 Using Pin 7 in digital mode
> INFO  | ??:??:?? 1 Using Pin 0 in PWM mode

What about if you also `--set device.buzzer_gpio 7`?

sdhuk, 2024-03-23:
That works! Thank you!

closed 2024-03-23 by sdhuk

todd-herbert, 2024-03-23:
@garthvh Is it feasible to refactor so that pwm also uses `external_notification.output_buzzer` instead of `device.buzzer_gpio`, or is there some reason why the two should remain separate?

ainformatico, 2025-07-08:
TL;DR:
You _have to user the CLI_ to set the `--set device.buzzer_gpio YOUR_GPIO`.
Just using the Android app to configure it does not work and makes the device crash and reboot.

Extended version:

I am using an active buzzer on the GPIO 48, used the Android app to set the buzzer GPIO to 48 and enable PWM. This resulted in the device just crashing and rebooting after receiving a message.

After carefully reading this GitHub issue, I decided to try using the CLI in addition to the settings I had done on the Android app.

```
--set device.buzzer_gpio 48
```

This made it work!

Looking at [this comment](https://github.com/meshtastic/firmware/issues/3436#issuecomment-2016515109) gave me the hint that there are two settings that might be clashing on the Android app.