meshtastic/firmware #3438: [Feature Request]: Don't Save MQTT Nodes to the NodeDB if IgnoreMqtt is on

garthvh, 2024-03-19:
### Platform

NRF52, ESP32

### Description

Right now my router node is set to ignore MQTT to not annoy my many nearby lora friends, but it still collects MQTT nodes in the nodedb.

tavdog, 2024-03-19:
Why have mqtt enabled if you want to ignore everything ?

garthvh, 2024-03-19:
Seems to be honoring the ignore mqtt for rebroadcasting, but not the node infos

tavdog, 2024-03-19:
the from_mqtt flag must not be getting set somewhere in the network.  In my test I just had client bridge on and tested by sending packets and nodinfo via python client. It ignored all packets including nodeinfos.  But when I had other nodes possibly repeating nodeinfos they did make it in and I could see that the from_mqtt flag wasn't set.

closed 2024-03-31 by garthvh

zapewne, 2024-05-24:
From my observations, if you disable MQTT then your node will not appear on the map.