meshtastic/firmware #3441: [Bug]: had trouble register board to the android app

Sheng2216, 2024-03-19:
### Category

WiFi

### Hardware

Other

### Firmware Version

meshtasticd_2.2.24.e6a2c06_arm64.deb

### Description

I am trying to set up an internal evaluation SX1280 board developed by RAKwireless. We have a Pi-HAT to mount the SX1280 onto the Raspberry Pi, so we are using the default pin definition defined in the config.yaml file (only uncommenting the section for SX1280). We have connected the power, SPI MOSI, SPI MISO, SPI SCLK, and SPI CE pins, as well as the busy, reset, and irq pins. It is important to note that I connected SX1280_DIO1 of the chip to the irq (GPIO 16) pin of the Pi. However, I am experiencing difficulties registering the board in the Android app. Basically, what I did was enter the Pi's IP address (I tried wifi and ethernet) in the app, but it never detect the Pi and SX1280. 

Later, I switched to the Waveshare SX1262 XXXM LoRaWAN/GNSS HAT, and I encountered the same issue. However, I was able to successfully register the Waveshare SX1262  HAT to the app previously. I wonder what might have gone wrong. Any help would be greatly appreciated!

### Relevant log output

```Shell
rak@raspberrypi:~/meshtastic $ sudo meshtasticd -c config.yaml 
Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using config.yaml as config file
B@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (91 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (114 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 Using ( as I2C device.
INFO  | ??:??:?? 0 Scanning for i2c devices...
INFO  | ??:??:?? 0 Scanning for i2c devices...
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.2.24.e6a2c06
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 22
/root/.portduino/default/prefs/module.proto does not exist
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
/root/.portduino/default/oem/oem.proto does not exist
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
WARN  | ??:??:?? 0 Using nodenum 0xddcac757 
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 13, using LORA_24
DEBUG | ??:??:?? 0 region=13, NODENUM=0xddcac757, dbsize=1
DEBUG | ??:??:?? 0 Read RTC time as 1710832821
DEBUG | ??:??:?? 0 Running without GPS.
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
/root/.portduino/default/prefs/cannedConf.proto does not exist
INFO  | ??:??:?? 0 No /prefs/cannedConf.proto preferences found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
DEBUG | ??:??:?? 0 Attempting to activate sx1280 radio on SPI port /dev/spidev0.0
WARN  | ??:??:?? 0 SX128xInterface(cs=21, irq=16, rst=18, busy=20)
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=812, sf=11, cr=4/5) packet symLen=2 ms, payloadSize=0, time 71 ms
DEBUG | ??:??:?? 0 (bw=812, sf=11, cr=4/5) packet symLen=2 ms, payloadSize=253, time 650 ms
INFO  | ??:??:?? 0 Radio freq=2404.469, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=LORA_24, name=LongFast, config=0, ch=5, power=10
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 2400.000000 -> 2483.500000 (83.500000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 102 x 812.500kHz
INFO  | ??:??:?? 0 Radio channel_num: 6
INFO  | ??:??:?? 0 Radio frequency: 2404.468750
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Set radio: final power level=10
INFO  | ??:??:?? 0 SX128x init result 0
INFO  | ??:??:?? 0 Frequency set to 2404.468750
INFO  | ??:??:?? 0 Bandwidth set to 812.500000
INFO  | ??:??:?? 0 Power output set to 10
INFO  | ??:??:?? 0 SX1280 Radio init succeeded, using SX1280 radio
INFO  | ??:??:?? 0 API server listening on TCP port 4403
DEBUG | ??:??:?? 0 (bw=812, sf=11, cr=4/5) packet symLen=2 ms, payloadSize=237, time 605 ms
DEBUG | ??:??:?? 0 LoRA bitrate = 391.735535 bytes / sec
INFO  | ??:??:?? 0 PowerFSM init, USB power=1
DEBUG | ??:??:?? 0 Enter state: BOOT
INFO  | ??:??:?? 30 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 30 [NodeInfoModule] sending owner !ddcac757/Meshtastic c757/c757
DEBUG | ??:??:?? 30 [NodeInfoModule] Initial packet id 1804289383, numPacketId 4294967295
DEBUG | ??:??:?? 30 [NodeInfoModule] Update DB node 0xddcac757, rx_time=0
DEBUG | ??:??:?? 30 [NodeInfoModule] handleReceived(LOCAL) (id=0x6b8b4569 fr=0x57 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 30 [NodeInfoModule] localSend to channel 0
DEBUG | ??:??:?? 30 [NodeInfoModule] Add packet record (id=0x6b8b4569 fr=0x57 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] Expanding short PSK #1
DEBUG | ??:??:?? 30 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] Installing AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] enqueuing for send (id=0x6b8b4569 fr=0x57 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] txGood=0,rxGood=0,rxBad=0
```

caveman99, 2024-03-19:
The Pi Firmware is still very new. Bluetooth Communication with apps is not supported, only the TCP Socket and the HTTP endpoints (if you enable the webserver) work. I am not sure if the android app can access the TCP socket, but the web endpoint should work if you enter the IP of the device.

caveman99, 2024-03-19:
you can always use the python CLI on the same device if you tell it to use 'localhost' to communicate

Sheng2216, 2024-03-19:
> you can always use the python CLI on the same device if you tell it to use 'localhost' to communicate

Thank you so much for the information. I followed your suggestion and ran the `meshtasticd `on one terminal of the same machine. On another terminal, I used `meshtastic --host localhost`. However, the Python CLI still encountered trouble finding the SX1280 chip. Please review the log provided below:

> rak@raspberrypi:~ $ ip a
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
>     inet6 ::1/128 scope host noprefixroute 
>        valid_lft forever preferred_lft forever
> 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
>     link/ether d8:3a:dd:ca:c7:54 brd ff:ff:ff:ff:ff:ff
>     inet 10.2.13.162/24 brd 10.2.13.255 scope global dynamic noprefixroute eth0
>        valid_lft 520sec preferred_lft 520sec
>     inet6 fe80::2c57:2ed3:b061:df94/64 scope link noprefixroute 
>        valid_lft forever preferred_lft forever
> 3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
>     link/ether d8:3a:dd:ca:c7:56 brd ff:ff:ff:ff:ff:ff
>     inet 192.168.230.140/24 brd 192.168.230.255 scope global dynamic noprefixroute wlan0
>        valid_lft 3220sec preferred_lft 3220sec
>     inet6 fe80::ba32:23a2:1d3a:63b1/64 scope link noprefixroute 
>        valid_lft forever preferred_lft forever
> 4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
>     link/ether 02:42:c5:a2:79:13 brd ff:ff:ff:ff:ff:ff
>     inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
>        valid_lft forever preferred_lft forever

> rak@raspberrypi:~ $ netstat -an | grep 4403
> tcp        0      0 0.0.0.0:4403            0.0.0.0:*               LISTEN     

> rak@raspberrypi:~ $ cd meshtastic/

> rak@raspberrypi:~/meshtastic $ source .env/bin/activate

> (.env) rak@raspberrypi:~/meshtastic $ meshtastic --host localhost
> Error connecting to localhost:Timed out waiting for connection completion

> (.env) rak@raspberrypi:~/meshtastic $ meshtastic --host 127.0.0.1
> Error connecting to 127.0.0.1:Timed out waiting for connection completion

> (.env) rak@raspberrypi:~/meshtastic $ netstat -an | grep 4403
> tcp        1      0 0.0.0.0:4403            0.0.0.0:*               LISTEN     
> tcp        0      0 127.0.0.1:46622         127.0.0.1:4403          FIN_WAIT2  
> tcp       43      0 127.0.0.1:4403          127.0.0.1:46622         CLOSE_WAIT

Sheng2216, 2024-03-20:
Thank you for the information. So far, I have been trying to test with the Waveshare SX1262 XXXM LoRaWAN/GNSS HAT. I have also attempted to start the web server on the same machine using the [webserver](https://github.com/meshtastic/web):

```
rak@raspberrypi:~/web $ pwd
/home/rak/web
rak@raspberrypi:~/web $ sudo pnpm dev

> meshtastic-web@2.2.19-1 dev /home/rak/web
> vite --host

  VITE v5.1.6  ready in 1539 ms

  ➜  Local:   http://localhost:5173/
  ➜  Network: http://10.2.13.164:5173/
  ➜  Network: http://10.2.21.147:5173/
  ➜  press h + enter to show help
```
then I start the meshtasticd with this config:

```
### Define your devices here using Broadcom pin numbering
### Uncomment the block that corresponds to your hardware
---
Lora:
  Module: sx1262  # Waveshare SX126X XXXM
  DIO2_AS_RF_SWITCH: true
  CS: 21
  IRQ: 16
  Busy: 20
  Reset: 18

  spidev: spidev0.0

Logging:
  LogLevel: debug # debug, info, warn, error

Webserver:
  #Port: 443 # Port for Webserver & Webservices
  Port: 5173
  #RootPath: /usr/share/doc/meshtasticd/web # Root Dir of WebServer
  RootPath:  /home/rak/web
```
Notice that I tested with port 5173, I also tried the default 443 port. However, when I try to connect the device from the web client, it hangs there forever. Please check the image below. I did notice that there are incoming API requests in the Meshtasticd log.
[image]
[image]

jp-bennett, 2024-04-04:
> vite --host
> 
>   VITE v5.1.6  ready in 1539 ms
> 
>   ➜  Local:   http://localhost:5173/
>   ➜  Network: http://10.2.13.164:5173/
>   ➜  Network: http://10.2.21.147:5173/
>   ➜  press h + enter to show help

When using the meshtasticd webserver, you don't run vite. meshtasticd will serve the full webapp. And it doesn't connect to port 4403, it's an API over HTTP/HTTPS. 

The android app should be able to talk to the PI over TCP, I use it that way myself. 

There were some fixes in the latest versions. You may have better success with 2.3.2

closed 2024-07-07 by garthvh