meshtastic/firmware #3445: [Bug]: Heltec V3: Reboot Loop on JSON pull from MQTT

rFronteddu, 2024-03-19:
### Category

BLE

### Hardware

Heltec V3

### Firmware Version

2.2.24.e6a2c06 Beta

### Description

I have two phones and two heltec v3. I can send a message from either phone and the other phone pushes the message through mqtt. If I push a message such as:

{"from":1,"type":"sendposition","payload":"test"}  to topic  msh/2/json/mqtt/

I either never see the message being pushed to MQTT or the device that pulls from MQTT  enters a reboot cycle until I flash it again.

Am I doing something wrong or is there a bug?

asjmcguire, 2024-03-19:
Perhaps try sending something valid to MQTT?

{    "from":actual-real-node-id
    "type":"sendtext",
    "payload":"this is a test"
 }

You have to ensure you have a channel called mqtt on the device too.

rFronteddu, 2024-03-19:
My bad, I changed the type to sendtext by hand but it was correct in my example. I did not set the channel mqtt in the device, so that may be the reason.

asjmcguire, 2024-03-19:
Channel is optional, but without it it will be sent to the mqtt channel, which other nodes might not have. Setting it to 0 will at least send it to primary channel so you can see the message on your other devices. From is the decimal id of the node you want to send the message from.

rFronteddu, 2024-03-19:
I made it! 

I wish there was some debug log that would tell me when I am making something funky, I am not sure exactly what was happening but the device behavior ranged from continuous reboot to unresponsiveness. Flashing from scratch and following your suggestion worked.

rFronteddu, 2024-03-19:
I thought that I had fixed the issue but I was tricking myself because I was consuming from the topic I was publishing in...

I am not sure what is going wrong. Yesterday I was able to have a message sent from one device to the other through the chat app, but whenever I add the push from mqtt it seems one of the two devices eventually freezes and I have to flash it to reset its configuration. In this second configuration I am not able to see any message being passed. If I connect to mqtt I do see the device advertising that it is alive but never the message I am trying to exchange. I assume I must be doing something wrong but I am having an hard time figuring out what.

rFronteddu, 2024-03-20:
I am starting to get more confident with my assumption that the firmware has some problem.

I did another round of tests with two different devices. 

Device A was used to send messages using the text app. 

Device B was configured as uplink towards meshtastic with default configuration other than changing the  mqtt root. 

Upon sending a message from A to B, I have seen the message using mqtt explorer which means that the message was correctly delivered to the mqtt broker.

The problem is that after a couple of minutes the system became unstable and messages were not reliably transmitted from A to B anymore. Once the system became unstable, I was not able to see  any messages reaching the MQTT broker. 

Disabling MQTT in B restores the ability of the two nodes to communicate reliably.

GUVWAF, 2024-03-20:
Please don't use the public server with default channel for this, there is just too much traffic that the devices cannot handle as pointed out in the [documentation](https://meshtastic.org/docs/configuration/module/mqtt/#connect-to-the-default-public-server).

rFronteddu, 2024-03-20:
I did change the root to msh/us/rob to avoid that problem and there was only me generating messages on that topic.

GUVWAF, 2024-03-20:
Please check the serial logs of Device B to see what's happening.

rFronteddu, 2024-03-20:
I will play again with it tomorrow. I was using the client proxy! I will try using the device WiFi and let you know!

closed 2024-03-26 by thebentern

rFronteddu, 2024-03-26:
I am waiting to get an hotspot to try and connect directly because the heltec doesn't seem to be getting a valid IP from my enterprise wifi. Will write in a few days.

rFronteddu, 2024-03-27:
I was able to make it work using a wifi hotspot. Not sure why the device was not working with my enterprise WiFi but I assume it's some unaccounted security policy. There are also several stability issues because  I had to reboot the heltec and set mqtt to on several times before the configuration was finally accepted.