meshtastic/firmware #3447: [Bug]: Issue Receiving certain messages

grahamrobertslx, 2024-03-19:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.24

### Description

When sending messages to my other node i keep getting the payload \020\224\251\331\322\r which gives me  message of ???? in diamond boxes. One node is on 2.2.24 and the other on 2.3.0. Both are T-Beams
[image]
[image]

paoloponzano, 2024-03-20:
Happening to me too

andrekir, 2024-03-21:
we're tracking this in #3452, please add any additional details or follow the progress there.

closed 2024-03-21 by andrekir